beagleboard/librobotcontrol #16: BeagleBone Green Wireless (wl18xx IRQ pin) and battery_monitor

RobertCNelson, 2016-12-15:
@StrawsonDesign & @jadonk
Please update the "battery_monitor" to check `cat /proc/device-tree/model` and definitely blacklist the BeagleBone Green Wireless, (to be safe, only run on the Blue/Blue-cape), as it takes away the IRQ pin we use on the wl18xx:

```
[   58.653762] genirq: Failed to request resources for wl18xx (irq 54) on irqchip 44e07000.gpio
```

https://github.com/StrawsonDesign/Robotics_Cape_Installer/tree/master/battery_monitor_service/battery_monitor

(background: we like to try using one image for all boards, so while battery_monitor is only for the blue/blue cape, it's enabled by default..)

Regards,

StrawsonDesign, 2016-12-15:
> Installer/tree/master/battery_monitor_service/battery_monitor

Noted

StrawsonDesign, 2016-12-16:
closed in testing for v0.3.0

closed 2016-12-16 by StrawsonDesign