beagleboard/librobotcontrol #21: Handle latest Blue device tree

jadonk, 2017-01-30:
The assumption that GPIO pins are available should be avoided and the case they aren't available should be handled. Memory mapping GPIO pins may typically still work, but the helper may or may not be available. Pinmux should be checked to see if the pin is in GPIO mode.

```
root@beaglebone:/var/lib/cloud9# test_motors -d 0.1
gpio/direction: No such file or directory
ERROR: Failed to set gpio pin 66 as output
gpio/direction: No such file or directory
ERROR: Failed to set gpio pin 67 as output
WARNING: Failed to configure all gpio pins
ERROR: failed to configure GPIO
failed to initialize cape
```

jadonk, 2017-02-01:
Quick hack: 78a2d6c0da04463289b0d18f0ccf8a2ceea5baf3

StrawsonDesign, 2017-02-04:
should be fixed in 0.3.0, just published https://github.com/StrawsonDesign/Robotics_Cape_Installer/releases/tag/v0.3.0

closed 2017-02-05 by StrawsonDesign