beagleboard/librobotcontrol #33: Improve "Buy Now" from the website

csobrinho, 2017-03-16:
It would be great if we could acquire a Robotics Cape as soon as possible but since it doesn't seem possible yet, why not add a Mailing List or a Contact Me page so that we can subscribe and know when the cape is available?

Also, couldn't find any contact/email on the website...

Thanks

dlech, 2017-03-16:
The BeagleBone Blue was just released this week. https://beagleboard.org/blue

csobrinho, 2017-03-20:
I'm was interested in the Robotic Cape for the BBBlack, not the new Beaglebone Blue

StrawsonDesign, 2018-04-13:
updated. It's available at http://www.newark.com/element14/bb-cape-robotics/robotics-cape-9-18-vdc/dp/95Y0637 although newark's website is down right now......

closed 2018-04-13 by StrawsonDesign