beagleboard/librobotcontrol #36: request: add ability to have multiple processes running RC libraries at the same time

finger563, 2017-03-27:
It would be nice to be able to have multiple processes (not just threads of the same process) that can interact with the hardware using the RC libraries.  This would be useful in systems where process level fault-mitigation strategies (such as restarting processes) allow for greater separation and componentization of code.  The same can be true for distributed systems process migration.

jadonk, 2017-03-29:
@finger563 can you provide specific suggestions on how to do that as well as achieve @StrawsonDesign goals of avoiding reuse of multiple of his example applications at the same time? I think if we had a way to specify greater granularity as an option or even sometimes invoke asynchronous kills for applications that don't want to leverage the get_state() loop, it would be of some benefit to general users.

finger563, 2017-03-29:
@jadonk I've been looking through the library code in the repo to see if I can start making a modification to what I'd like and then maybe send a PR.  The code we're working with doesn't use any of the state functionality that RC provides, so I'm just looking into how it would actually be used as a library properly instead of requiring a singleton as it currently does.  Is there a design doc anything for the internals of the library other than just reading through the code and the (nicely detailed) comments?

finger563, 2017-03-31:
Correct me if I'm wrong, but it seems the only needed feature is the state functionality that is used within the libraries.  If I remove or make optional the dependence on the `rc_get_state()` function then I should be able to make much of the library usable across processes, correct?

StrawsonDesign, 2017-03-31:
Hi finger563, I'm trying to achieve this with the v0.4.0 branch right now. rc_initialize() will be deprecated and each subsystem will need its own initialize and cleanup functions which is less clean than I wanted it to be, but on the upside each subsystem can be compiled and run independently. the rc_get_state() and rc_set_state() functions will also not have any effect on library function, they will just exist to assist in the flow of multithreaded programs if you want to use them. Right now I'm pretty sure the IMU, barometer, timing, math, and LED functions can run independently without rc_initialize().

finger563, 2017-04-03:
@StrawsonDesign, thanks that' what I'd found in my inspection as well.  What about the ADC/GPIO libraries requires the initialization / state functionality?

StrawsonDesign, 2017-04-03:
ADC and the GPIO functions with 'mmap' in the function name use direct memory access and are currently interdependent only due to a shared memory map pointer created during rc_initialize. Splitting those and making them independently is on my shortlist for v0.4.0. references to the rc_state are very minimal inside the library, it's really just used to make sure threads exit if state becomes set to EXITING, removal of this dependency will mean many sections will require a stop function to make sure threads stop cleanly. These stop functions already exist for imu/barometer/dsm but are not strictly necessary right now if the user sets rc_state to exiting.

closed 2018-04-13 by StrawsonDesign