beagleboard/librobotcontrol #40: i I executed the examples of rc_balance,but the motors cannot work

leehua1993, 2017-05-07:
who can help me?
debian@beaglebone:/$ rc_balance
Inner Loop controller D1:
order: 2
timestep dt: 0.0100
-4.9450z虏 +  8.8620z  + -3.9670
--------------------------------
 1.0000z虏 + -1.4810z  +  0.4812

Outer Loop controller D2:
order: 2
timestep dt: 0.0100
 0.1886z虏 + -0.3721z  +  0.1835
--------------------------------
 1.0000z虏 + -1.8605z  +  0.8605

Hold your MIP upright to begin balancing

dsm Calibration File Doesn't Exist Yet
Run calibrate_dsm example to create one
Using default values for now

RUNNING: Hold upright to balance.
    胃    |  胃_ref  |    蠁    |  蠁_ref  |    纬    |  D1_u   |  D3_u   |  vBatt  |arm_state|
  0.216  |  0.000  |  0.213  |  0.000  |  0.006  |  1.000  | -0.008  |  8.603  |  ARMED  |inner loop controller saturated
  0.362  |  0.025  |  0.359  |  0.000  |  0.006  |  1.000  | -0.008  |  8.613  |DISARMED |

the project version is 0.3.4. I execute the rc_test_motors -d0.5.the two motors can run as usual.
I  may think if it is the PID date is not confihuration.who can help me.

StrawsonDesign, 2017-05-12:
Are the motors plugged into channels 2&3? Older copies of the eduMiP instructions may be floating around which say channels 1&4. running rc_test_motors will drive all 4 channels, not just the two that rc_balance will try to drive. I have new instructions here with the new correct headers: http://strawsondesign.com/#!edumip-assembly
I hope it's just the motor connections.

jadonk, 2017-05-24:
@leehua1993 can you confirm if you followed the instructions?

closed 2018-04-18 by StrawsonDesign