beagleboard/librobotcontrol #45: gcc command line compiling

hugoeverysens, 2017-05-30:
Hi,
I'm trying to start a project on the bbblue. I'm compiling on it directly.

At last, to make it easy, I try to compile the project through gcc/g++.

In the same directory, I have the main:
```
#include "rc_usefulincludes.h"
#include "roboticscape.h"
int main(){
	rc_initialize();
}
```
And : libroboticscape.so, rc_usefulincludes.h,roboticscape.h

I do try this:
`gcc -g  -o binary  -lroboticscape main.cpp`
and this is what happens:
```
In file included from main.cpp:6:0:
libraries/rc_usefulincludes.h:15:0: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE  // to enable macros in pthread
 ^
<command-line>:0:0: note: this is the location of the previous definition
/tmp/cc7mQL20.o: In function `main':
/home/debian/robot/main.cpp:14: undefined reference to `rc_initialize()'
collect2: error: ld returned 1 exit status
```

I'm a bit annoyed, at first because I don't like warnings, but also because, I checked, and rc_initialize is really in libroboticscape.so.

How can I compile in command line?
Do you know the other place where the define is defined? (where should I remove it?)

dlech, 2017-05-31:
Try adding `#ifndef _GNU_SOURCE` before `#define _GNU_SOURCE`

hugoeverysens, 2017-05-31:
thank you for your help! What about the inclusion of the shared library?

```
gcc -g -o binary -lroboticscape main.cpp
/tmp/cccwXOan.o: In function `main':
/home/debian/robot/main.cpp:14: undefined reference to `rc_initialize()'
collect2: error: ld returned 1 exit status
```
I checked and the function should be in it. Is it something wrong with my command line?

```
nm -g libroboticscape.so 
0000742c T rc_initialize
```

dlech, 2017-05-31:
If you are writing c++, then you need to add 

```
#ifdef __cplusplus
extern "C" {
#endif
```

and 

```
#ifdef __cplusplus
}
#endif
```

to the beginning and end of header files respectively.

You should make a pull request for these fixes if you get it working. :smiley:

hugoeverysens, 2017-05-31:
It works great! I might see later to pull the diff.
this solve all my problems, as at the end, I want to use c++
Also, now I can use better cmake, and it also work with this CMakeLists.txt:

```
cmake_minimum_required(VERSION 3.5.2)
project (robot)
find_library(roboticscape roboticscape libraries)
message("roboticscape is at: ${roboticslib}")
add_executable(robot main.cpp)
TARGET_LINK_LIBRARIES(robot roboticscape)
```
I hope it can help some beginners like me :)

closed 2017-05-31 by hugoeverysens

tabris2015, 2018-09-18:
> It works great! I might see later to pull the diff.
> this solve all my problems, as at the end, I want to use c++
> Also, now I can use better cmake, and it also work with this CMakeLists.txt:
>
> ``` [...]

Hi! I'm trying to do the same, do you have a sample project with the correct form for using this library with cmake?

RamonHA, 2018-10-17:
I am having a similar problem.
The code is the next:

#include <rc/i2c.h>
#include <iostream>

#define I2C_MAX_BUS 5
#define I2C_BUFFER_SIZE 128 

int bus=1;
uint8_t address=14;
uint8_t data;

int main(){
        rc_i2c_init(bus, address);
        std::cin>>data;
        rc_i2c_write_byte(bus, address, data);  
        return 0;
}

And it shows the sames mistakes as hugoeverysens.

i2c-test.cpp:(.text+0x18): undefined reference to `rc_i2c_init'
i2c-test.cpp:(.text+0x40): undefined reference to `rc_i2c_write_byte'

i2c-test.cpp   ----> Name of the c++ file
I try to follow the solution here was proposed, but in the header files there is already the:

#ifdef __cplusplus
extern "C" {
#endif

and the: 

#ifdef __cplusplus
}
#endif

-----------------------------------------------------------------------------------------------------------------
The only difference is that there is a begging #if and final #endif, yet they correspond to the header as it is.

I run it in terminal as "g++ i2c-test.cpp -o main"

StrawsonDesign, 2018-10-17:
@RamonHA you also need to link to the library, try:

g++ i2c-test.cpp -lrobotcontrol -o main