beagleboard/librobotcontrol #46: Occasionally hanging while reading jack voltage

aanon4, 2017-06-01:
In my app I periodically poll the battery and jack voltages. Very, very, occasionally my app will hanging while reading the jack voltage. This is the stack trace (sorry for the limited symbol information):

Thread 1 (Thread 0xb6c9d000 (LWP 5197)):
#0  0xb4c3c3d8 in mmap_adc_read_raw () from /usr/lib/libroboticscape.so
#1  0xb4c4126a in rc_adc_volt () from /usr/lib/libroboticscape.so
#2  0xb4c41304 in rc_dc_jack_voltage () from /usr/lib/libroboticscape.so
#3  0xb4c6dd68 in BBBluePower::JS_jackVoltage(v8::FunctionCallbackInfo<v8::Value> const&)
    ()
   from /root/borgcube-rov/hw/board/beagleboneblue/native/build/Release/beagleboneblue.node
#4  0x00403158 in v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&)) ()
#5  0x0042e290 in ?? ()

I experimented with a few breakpoints and have determined that mmap_adc_read_raw never returns. I see a few while loops in the code which - obviously - should never fail, but it appears that occasionally they do.

Thanks
Tim

aanon4, 2017-06-01:
Could the problem be because 'rc_battery_monitor' is running in the background at the same time?

StrawsonDesign, 2017-06-02:
interesting. mmap_adc_read_raw is a very simple function that reads directly from the adc registers. I haven't observed this in the past. Does stopping rc_battery_monitor (systemctl stop rc_battery_monitor) alleviate the hangup?

RobertCNelson, 2017-06-03:
Yeah, we "really" need to move that from mmap'ed to iio, then users can still dump the raw values from /sys/bus/iio/*

Regards,

aanon4, 2017-06-09:
Yes - once I stop it running rc_battery_monitor, the problem goes away. I'm assuming that very occasionally both the rc_battery_monitor code and my code are in those loops in parallel, both do the setup in parallel, but only one get the exit condition because it's shared.

clydemcqueen, 2017-10-13:
I'm also seeing this -- I'm reading at 50Hz, and it consistently happens in less than a minute. Turning off rc_battery_monitor solves it. My stack trace:

#0  0xb6eb74f0 in mmap_adc_read_raw () from /usr/lib/libroboticscape.so
#1  0xb6ebc3c2 in rc_adc_volt () from /usr/lib/libroboticscape.so
#2  0xb6ebc45c in rc_dc_jack_voltage () from /usr/lib/libroboticscape.so
#3  0x00033622 in orca_driver::OrcaDriver::publishVoltage() ()
#4  0x0003376c in orca_driver::OrcaDriver::spinOnce() ()
#5  0x00033ac0 in orca_driver::OrcaDriver::run() ()
#6  0x00033e86 in main ()

Thanks.

StrawsonDesign, 2018-04-13:
new release v0.4.0 uses IIO to read adc and problem has gone away.

closed 2018-04-13 by StrawsonDesign