beagleboard/librobotcontrol #48: Cannot use CAN1 bus

DWatkins, 2017-07-06:
Unless I am missing something, it does not seem that the CAN bus can be fully set up at this time.  

Device: Blue

Kernel: 4.4.39-ti-r75

rc version: 0.3.4

Looking at `BB-DCAN1-00A0.dts` we see the following two fragments required for setting up CAN1 if we did not have the  robotics cape library installed

```
/dts-v1/;
/plugin/;
 
/ {
    compatible = "ti,beaglebone", "ti,beaglebone-black";
 
    /* identification */
    part-number = "dcan1pinmux";
 
    fragment@0 {
        target = <&am33xx_pinmux>;
        __overlay__ {
            dcan1_pins_s0: dcan1_pins_s0 {
                pinctrl-single,pins = <
                    0x180 0x12  /* d_can1_tx, SLEWCTRL_FAST | INPUT_PULLUP | MODE2 */
                    0x184 0x32  /* d_can1_rx, SLEWCTRL_FAST | RECV_ENABLE | INPUT_PULLUP | MODE2 */
                >;
            };
        };
    };
 
    fragment@1 {
        target = <&dcan1>;
        __overlay__ {
             #address-cells = <1>;
             #size-cells = <0>;
 
             status = "okay";
             pinctrl-names = "default";
             pinctrl-0 = <&dcan1_pins_s0>;
        };
    };
};
```

`fragment@0` is responsible for setting the pin mux and can be handled readily either by using the robotic's cape built in pinmux functions 
```
rc_set_pinmux_mode(UART1_HEADER_PIN_3, PINMUX_CAN);
rc_set_pinmux_mode(UART1_HEADER_PIN_4, PINMUX_CAN);
```
or by directly setting the state of the pins in the file system 

```
echo can > /sys/devices/platform/ocp/ocp\:P9_24_pinmux/state 
echo can > /sys/devices/platform/ocp/ocp\:P9_26_pinmux/state 
```

But I do not currently see a way to perform the function of `fragment@1` to enable the CAN device after setting up the pinmux.

RobertCNelson, 2017-07-06:
@DWatkins the dcan1 node on uart0_rtsn.dcan1_rx/uart0_ctsn.dcan1_tx was enabled by default on r82

Feel free to upgrade

```
cd /opt/scripts/tools/
git pull
sudo ./update_kernel.sh
```

Regards,

DWatkins, 2017-07-06:
Thanks! That worked great.  

For reference if I forget or others need it:

after updating the new kernel version as specified by @RobertCNelson , set the pinmux as outlined in my first post. Then run 
```
sudo ip link set can0 up type can bitrate [bitrate]
sudo ifconfig can0 up
```

closed 2017-07-06 by DWatkins