beagleboard/librobotcontrol #49: DCan1 RX not read

DWatkins, 2017-07-27:
Continuing on configuring dcan as in #48 , it appears that dcanRX is not configured to read properly.

I connected the blue to a pre-existing can bus and viewed its output with a logic analyzer.

After configuring can0 to the proper bitrate, running `candump can0 -L` on the blue produces no data, even though traffic was seen on the logic analyzer.

WIth can0 configured with no loopback, writing to the bus fails, as shown below.

[image]

After enabling loopback, the blue can successfully write to the bus:

[image]

But it follows no arbitration and will write even if the bus is already in use. When the bus then goes to an error state, it ignores it and continues to write.

[image]

This is fairly consistent with no actual RX feedback, even though the logic analyzer tied to the RX and TX pins on the blue show proper connection.

DWatkins, 2017-08-08:
This appears to be a similar issue?

https://groups.google.com/forum/embed/?place=forum/beagleboard&showsearch=true&showpopout=true&showtabs=false&hideforumtitle=true&parenturl=https%3A%2F%2Fbeagleboard.org%2Fdiscuss#!category-topic/beagleboard/can/2fcUNqcMTXQ

jadonk, 2017-08-08:
Here's how I tested CAN on Blue: https://gist.github.com/jadonk/90d3462c8569d11fa3f7caf14f4769f1

DWatkins, 2017-08-08:
@jadonk 

Sadly setting up the can interface doesn't work for me. I can send data just fine after setting it up, but cannot receive anything.

StrawsonDesign, 2018-04-13:
closing since CAN is not part of this git repo.

closed 2018-04-13 by StrawsonDesign