beagleboard/librobotcontrol #50: rc_test_encoders  different count between CW -vs- CCW

patrickpoirier51, 2017-07-31:
Just received my BBBlue , upgraded the robotic cape  and I connected this Motor & Encoder:
https://www.banggood.com/6V-210RPM-Encoder-Motor-DC-Gear-Motor-with-Mounting-Bracket-and-Wheel-p-1044064.html?rmmds=buy

Testing with **rc_test_encoders** and  when turning the wheel manually, the display shows a persistent error between CW and CCW readings.
Example: 
Manually slowly turning the wheel CW 1/4 turn = its counting up to 180
Manually slowly turning the wheel CCW 1/4 turn = its counting down to -90
Same pattern on both motors-encoders.
I swapped phases  , No changes

I am available for further tests

closed 2017-08-06 by patrickpoirier51

jadonk, 2017-08-08:
@patrickpoirier51 did you figure out what's going on? Is that a good/bad motor/encoder pair to use?

patrickpoirier51, 2017-08-08:
@jadonk  It was related to the encoder that was really out-of spec == about 75 degree phase shift. I modified them so that they generate a clean quadrature signal.

Look here:
The bad signal: https://files.gitter.im/patrickpoirier51/BNOd/image.png
The modification : Orange is initial angle and green the angle once I cut the sensor plastic base realign with a pair of flat pliers
https://files.gitter.im/patrickpoirier51/8xeC/image.png