beagleboard/librobotcontrol #53: Strange Magnetometer Data

msardonini, 2017-08-22:
I've been seeing some strange magnetometer data coming out of the robotics cape library and my BB Blue. Here is a snippet of a log I took, which is just the "mag" field array within the "imu_data_t" structure. The recording is of one slow 360 degree spin (x axis is in seconds). 

The X-Direction seems to be behaving, Y is ~0 (not actually zero) and Z is nonsense. Perhaps the wrong bytes are being parsed from the i2c read?

I started investigating this when I noticed the fused yaw angle was becoming unstable, which using magnetometer heading to correct for gyro drift. Has anyone else seen similar behavior?

[image]

msardonini, 2017-09-12:
I've found a temporary fix for this issue:

Lines 1715 - 1717 can be changed from this:
		mag_adc[0] = (int16_t)(((int16_t)raw[i+1]<<8) | raw[i+0]);  
		mag_adc[1] = (int16_t)(((int16_t)raw[i+3]<<8) | raw[i+2]);  
		mag_adc[2] = (int16_t)(((int16_t)raw[i+5]<<8) | raw[i+4]);
to this:
		mag_adc[0] = (int16_t)(((int16_t)raw[i+1-4]<<8) | raw[i+0-4]);  
		mag_adc[1] = (int16_t)(((int16_t)raw[i+3-4]<<8) | raw[i+2-4]);  
		mag_adc[2] = (int16_t)(((int16_t)raw[i+5-4]<<8) | raw[i+4-4]);

It doesn't look like the "int i" here is properly set before entering these statements. In my application, it is off by 4 bytes.

dangwalsh, 2017-09-16:
This seemed to fix my issues with calibrating the magnetometer.  Thanks!

OscarJGV26, 2017-09-19:
I am getting a difference issue. Perhaps you guys can verify it.

I configured imu for dmp with magnetometer enabled at 50 Hz.

My main loop is also running at 50Hz using timer SIGRTMIN which I use to call a function that simply changes a valuee from 0 to 1 so that main loop runs.. 

And I am getting this strange reading periodically, 

Euler = -0.2870, -1.4358, 0.0247, Gyro = 0.0610, -0.0610, -0.0000, Accel = 0.2406, 0.0353, 9.9575, Mag = 4872.4038, 67.2848, 6387.8413, Quat = 0.9999, 0.0125, -0.0025, -0.0002, 
Receiver Malfunctioning... Values are outside range!
Your receiver seems to have disconnected!
Euler = -0.2857, -1.4346, 0.0241, Gyro = 0.0610, 0.0610, -0.0000, Accel = 0.2424, 0.0239, 9.9270, Mag = 9.1489, -26.1872, -63.2956, Quat = 0.9999, 0.0125, -0.0025, -0.0002, 
Receiver Malfunctioning... Values are outside range!
Your receiver seems to have disconnected!
Euler = -0.2844, -1.4343, 0.0235, Gyro = 0.0000, -0.0000, -0.0610, Accel = 0.2508, 0.0539, 9.9264, Mag = 9.1489, -25.5880, -61.5641, Quat = 0.9999, 0.0125, -0.0025, -0.0002,

msardonini, 2017-09-19:
It looks like your strange magnetometer readings are similar to what I observed, notice that the graph above has it's y-axis values scaled by 10^4. That problem was parsing the wrong bytes within the buffer "raw". 

It's interesting how you aren't getting bad data consistently like I saw. Perhaps the value of "i" isn't consistent when it reaches the statements I highlighted above?

Note that you need to run "make install" from the libraries folder in order for any of those changes to take effect.

OscarJGV26, 2017-09-19:
Hi @msardonini 
Thank you for your prompt response. 

I can confirm that my magnetometer is working perfectly fine off-the-shelff  (without any modification to the library) WHEN I don't have dmp feature, meaning when I am reading the sensor by myself (one-shot).

Indeed, I think it has something to so with the bytes being concatenated strangely but I wouldn't know for sure. For now, I will just leave the dmp feature and process the data fusion outside by myself since I've found another deal breaker in (https://github.com/StrawsonDesign/Robotics_Cape_Installer/issues/74) just in case you wanted to check that out. 

Regards,
Oscar

ycdtosa, 2017-11-20:
We run into this same problem, and @msardonini solution is working for us.
Still i have to take a look why do we need this change, and where comes the i index from.

sleven79, 2018-02-03:
The 4 byte offset due to variable "i" is not surprising. The value of this variable is overwritten some lines above (in the part on quaternion normalization) where "i" is used in some for loops.

`		// do double-precision quaternion normalization since the numbers
		// in raw format are huge
		for(i=0;i<4;i++) q_tmp[k]=(double)quat[i];
		sum = 0.0;
		for(i=0;i<4;i++) sum+=q_tmp[i]*q_tmp[i];
		qlen=sqrt(sum);
		for(i=0;i<4;i++) q_tmp[i]/=qlen;
		// make floating point and put in output
		for(i=0;i<4;i++) data->dmp_quat[i]=(float)q_tmp[i];`

Instead of correcting your offset in the mag_adc[x] calculation, replace "i" in the for loops by another counter variable.

ycdtosa, 2018-02-04:
IIRC that is what i did...
let me take a look to the PR

ycdtosa, 2018-02-04:
yeap...
this was the PR comment:

> small loops where reseting the i variable,
> used as index in the FIFO, so i just used
> a different variable for the loop.

thus that is very likely how i fixed it.

StrawsonDesign, 2018-04-13:
fixed in v0.4.0

closed 2018-04-13 by StrawsonDesign