beagleboard/librobotcontrol #59: Add error checking to "rc_left/right_multiply_matrix_inplace()"

jselfridge, 2017-08-29:
Within rc_matrix.c, in functions rc_left/right_multiply_matrix_inplace():
Because these two functions only work on square matrices, there should be some minor error checking, to confirm that both matrices A and B are square, and that they both share common dimensions.

jselfridge, 2017-08-29:
Correction.  They don't both have to be square at the same time; but one must be square, and the other must have very specific dimensions.  Consider the following dimensions: A(nxm) B(mxp) = C(nxp).  For the most general case, n!=m!=p, C will not fit in either A nor B.  For the case when n=m, then A is square, and C can fit within B.  Similarly, when m=p, then B is square, and C can fit within A.

StrawsonDesign, 2017-10-13:
I added a dimension check to make sure A.cols==B.rows, but only so the warning message printed to the screen shows the higher level function name instead of "rc_multiply_matrices" which would have caught that later. In the left/right 'inplace' functions the result of the multiplication is still placed in newly allocated memory and the matrix struct A or B is then freed and directed to the new memory. Therefore C does not technically need to fit in the memory allocated by A or B. Or perhaps I misread your request?

closed 2018-04-13 by StrawsonDesign