beagleboard/librobotcontrol #60: Request on Servo/ESC library: Allow Pulses of certain length with certain Frequency

OscarJGV26, 2017-08-30:
I would like to suggest the Servo/ESC functions to modified so that they can generate pulses of certain length and certain frequency and not just one pulse of certain length.

As it is said in the library, ESCs and even digital servos allow update rates of up to 400 Hz. However, you don't give a way to define this frequency INDEPENDENTLY of whatever you are running in your main program. 

Your function rc_test_servos uses function rc_usleep to achieve the update rate, but what if you don't wanna have to be updating at 400 Hz on the main loop? Clearly you want to be updating as fast as you can to overcome the filtering of ESCs which you can see here (https://www.rcgroups.com/forums/showthread.php?1250488-ESC-Response-Testing). However, that doesn't mean that you necessarily need to be taking decisions at the same rate. If your main loop is running at 50 Hz, the difference in the update rate can have a significant impact on the flight performance.

This could also be achieved by "timer interrupts" but since I am noob on linux I still don't know how to implement them in the beaglebone blue.

MarkAYoder, 2017-08-30:
I have examples of how to use the PRU to do both the frequency and pulse width of the PWM signal.[1]  I measured the PRU generating a very stable PWM at over 1 MHz, no need for the ARM to service it.

Unfortunately, there is some assembly required...

--Mark

[1] https://github.com/MarkAYoder/BeagleBoard-exercises/tree/master/pru/examples

OscarJGV26, 2017-08-30:
Hi Mark,

Thank you very much for your response.

I will definitely look into your examples. That is exactly what I need. A PWM, as precise and stable as possible, on all the outputs and I can work with 400 Hz, 1 MHZ is a little bitch too much for me. :P 

Can I ask you, what do I need to do to get one of your examples going? And, would uploading your examples affect the robotics libraries? 

I ask because I assume it would need to change the program of the prus and I need the program of the encoder to read my RC reciever. Maybe I can use both of your programs. I just need to read a 6 channel RC receiver with pulses between 596-1595. 

Regards,
Oscar

MarkAYoder, 2017-08-30:
If you look around the directories you'll see I have have install.sh and setup.sh files that give instructions on how to make it work.
Currently it's a 18 channel PWM, 6 on PRU0 and 12 on PRU1.  You can easily remove the channels you don't want.  

I think the robotcape runs the 4th encoder on PRU0 and the pwm channels on PRU1, so if you use PRU1 for the new pwm you can leave the encoder alone.

I hope this helps....

--Mark

OscarJGV26, 2017-08-30:
Dear Mark, 

Thank you for your response. 

I would like to ask you this.

On my Teensy 3.5, I generate the signals to the ESCs using a 12-bit resolution PWM running at 400 Hz to which I send signals between 1638 minimum (1 ms pulse width) and 3276 (2 ms pulse width). So as you can imagine, this has a very high resolution of 1638.

Now, If I understand correctly, I see that the pwm functions that your library has in the examples, e.g. start_pwm_us, have a duty cycle (on-time) of 0-100 (integer) which means I can only access barely a resolution of 40 (1 ms pulse at 400 Hz) to 80 (2 ms pulse at 400 Hz), which means your possible values are

1ms,1.025ms, 1.05 ms, 1.075ms, 1.1ms, and so on....

Am I correct? 

I would like to have more resolution. Would you say is possible to get a higher resolution? I need it to be at least 100 or if possible 1000. 

Regards,
Oscar

MarkAYoder, 2017-08-30:
Oscar:
  If you check in pwm-test.c[1] you'll see a routine called start_pwm_us() which let's you give the pulse width in micro seconds.  1us should get you plenty of resolution.  If not, the routine start_pwm_count() let's you specify how many counts to be on and off which gives you very fine control.

The range of 0 to 100 was just to make an easy interface.

Have you considered using the hardware PWM's an skip the PRUs? 

--Mark

[1] https://github.com/MarkAYoder/BeagleBoard-exercises/blob/master/pru/examples/pwm/pwm-test.c

OscarJGV26, 2017-08-30:
Yes, I was looking into the eHRPWM which has an outstanding 16 bit resolution, but I don't know how to connect it to the output pins in the beaglebone blue and it only has 3 PWMs, I need 5 (2 servos/3 ESCs). The servos could be controlled with lower resolution PWM. I am very new to linux. If you could give me a few tips on how to achieve this I would be grateful.

Nevertheless, I see now how you are doing the count-countOn.

I will try to do it that way.

Thank you very much for your help!

Regards,
Oscar

OscarJGV26, 2017-08-30:
Hey Mark, 

Since you developed the encoder library, I would like to ask you, how do I get to read CPPM signal from my receiver, i.e. all channels merged into one pin.

I fail to see how that is possible since you only increment/decrement encoder count but no where you declare channels. I was also thinking about the "DSM" library but I don't know if it can handle CPPM.

MarkAYoder, 2017-08-30:
Yes, there are 3 PWMs, but they each have 2 halves.  A pair of PWMs must have the same frequency, but can have different duty cycles.  On the Blue PWMs 1A, 1B, 2A and 2B drive the 4 motor h-bridges.  It looks like PWM0A and 0B are muxed with UART2.  So if you don't need the motors and UART2 you can have 6 PWMs.

OscarJGV26, 2017-08-30:
Yes. That's exactly what I found. But, can I re-direct the signals to use the on-board 3-pin outputs? I'd rather avoid having to change the servo cables which might end up messing up cabling a little.

closed 2017-09-02 by OscarJGV26

jadonk, 2017-09-05:
@MarkAYoder can you patch the code provided by @StrawsonDesign to still work with the roboticscape library, but provide any additional functions? I'd like to treat this project as an upstream of sorts.