beagleboard/librobotcontrol #61: Request on RC: Allow CPPM (or PPMsum) signals to be aqcuired.

OscarJGV26, 2017-08-30:
I would like to suggest a modification or option in the pru0 library to be able to identify the CPPM protocol used by many RC receivers. I am not sure weather this is possible using encoder library by itself and I am almost sure it needs a modification. But please, let me know if I missed anything.

jadonk, 2017-08-31:
@OscarJGV26 The firmware in ArduPilot for the PRU seems to do this. Perhaps you can seek out how that and the quadrature encoder firmware can be combined and provide a pull request?

OscarJGV26, 2017-08-31:
Hi Jason,

Yes I will probably be doing that. However I am unable to find the specific PRU firmware that runs on it. I'd appreciate if you know where I can find it. 

On other things, I wanted to ask you if you know anything about Connection refused 443. 

I am unable to apt-get update or git clone because of this.

Also, I am using Windows 10 64 bit and installed the drivers but there appears to be a problem in device manager. The beagle is seen in Ports as Gadget Serial (COM6) but with a warning saying this device cannot start. (Code 10). 

If you know anything about that, I'd appreciate it.

Nevertheless, I have already started coding using the roboticscape library for timing and imu, and is working perfectly.

Regards,
Oscar

OscarJGV26, 2017-09-01:
@jadonk Hi Jason, 

Did you see my message? I need help with this, please. 

I can't git clone anything or apt-get update. I just ran the beagle with a new flash image given in (https://github.com/beagleboard/beaglebone-blue/wiki/Frequently-Asked-Questions-(FAQ)#What_system_firmware_should_I_use_for_starting_to_explore_my_BeagleBone_Blue) and it is still refusing the connection on port 443.

If you have any idea what could bee wrong please let me know!

Regards,
Oscar

OscarJGV26, 2017-09-07:
@jadonk 

I've managed to use the PRUs of ardupilot instead of RC. Maybe this is of use for you? 

The code shown in https://github.com/OscarJGV26/rcout, particularly RC_Out.h/RC_Out.cpp and you are able to use the PRUs completely outside of ardupilot...

I am working on RC_In.h/RC_In.cpp for receiving the signals from my receiver and will be updating you when I get that working...

StrawsonDesign, 2018-04-13:
Recently the PRU driver updated to allow multiple remoteproc binaries to exist and be independently loaded from userspace so having different binaries for encoder and PRU delectable at runtime is easy now. If you can give me a PRU0 binary that works on an encoder PIN I'd happily integrate this in.