beagleboard/librobotcontrol #63: Function "rc_normalize_quaternion_array" has two bugs

jselfridge, 2017-09-02:
Within "quaternion.c", and within function "rc_normalize_quaternion_array", the for loops only apply to the first three elements within the quaternion.

currently:
    for(i=0;i<3;i++) sum+=q[i]*q[i];
    for(i=0;i<3;i++) q[i]=q[i]/len;
revision:
    for(i=0;i<4;i++) sum+=q[i]*q[i];
    for(i=0;i<4;i++) q[i]=q[i]/len;

StrawsonDesign, 2017-09-02:
Justin, thank you so much for finding these bugs in the math library. I haven't had anyone look over my shoulder as I put this together, let alone do a proper code review so I am grateful for your keen eye. I'm getting married this weekend but will be back in the office shortly to merge these into the bugfixes branch. After a few other tweaks these fixes will be published as a new version.

jselfridge, 2017-09-02:
Perfect.  I've been very impressed with this API overall.  It's very clean, concise, and well organized.  I put my own Matlab-like library on the back burner while completing my PhD, and once I started back into hardware mode, I realized that utilizing this library would expedite my current research project.  I have some other functions for EKF, 9DOF AHRS, and GPS functionality, I'l pass your way once I'm back up to speed.

Congrats on getting married.  Delegate everything you can to other people and enjoy the day... it flies by so fast.

StrawsonDesign, 2017-10-13:
fixed in commit 909fe976719e92baa41c2c00cd1fbb9ea1945cf4

closed 2017-10-13 by StrawsonDesign