beagleboard/librobotcontrol #68: Issue with "rc_alloc_filter_from_arrays".

jselfridge, 2017-09-12:
Within file "rc_filter.c", within function "rc_alloc_filter_from_arrays".

Both the arrays for numerator and denominator initialize to the same length, no matter what is passed into the function.  The problem is that some functions will later require the relative degree of the transfer function to work properly.  But if the TF is formed with this function, it will always have a relative degree of exactly zero, which is not always accurate.

Current code:
	if(unlikely(rc_vector_from_array(&f->num,num,order+1))){
		fprintf(stderr,"ERROR in rc_alloc_filter_from_arrays, failed to alloc vector\n");
		return -1;
	}
	if(unlikely(rc_vector_from_array(&f->den,den,order+1))){
		fprintf(stderr,"ERROR in rc_alloc_filter_from_arrays, failed to alloc vector\n");
		rc_free_vector(&f->num);
		return -1;
	}

jselfridge, 2017-09-12:
Also, the error checking should be the same between "rc_alloc_filter_from_arrays" and "rc_alloc_filter".

StrawsonDesign, 2017-10-12:
fixed in v0.4.0 commit 96f388d8b66fe7439b0a102d9370fd84c400c95e

closed 2017-10-12 by StrawsonDesign