beagleboard/librobotcontrol #74: Dmp interrupt feature not giving deterministic reading times

OscarJGV26, 2017-09-19:
I found something that unfortunately is not going to work with my system given that I assume discrete/constant sampling time. Maybe for your case you can handle it in a different way but for me, this is a deal breaker. Sooo... here is the problem.

I have a function (Main) which runs at 50 Hz using timer interrupts/handlers. If I run this function with chrt -r 99 ./Main, this is the accuracy of the period I am getting which I think is enough (not gonna complain).
disconnected!
Real Period = 0.020015
Real Period = 0.020010
Real Period = 0.019971
Real Period = 0.019990
Real Period = 0.020033

I measure this period by saving the time with,
clock_gettime(CLOCK_REALTIME,&real_time_clock);

The interrupt function that I set for the dmp, saves another time which is approximately when the imu is read...
void imu_handler(){
    clock_gettime(CLOCK_REALTIME,&real_time_clock_imu);
}

Then I print the difference....
 rt_clock=real_time_clock.tv_sec+(double)real_time_clock.tv_nsec/1000000000.0f;
rt_clock_imu=real_time_clock_imu.tv_sec+(double)real_time_clock_imu.tv_nsec/1000000000.0f;
printf("Time difference = %lf\n",rt_clock-rt_clock_imu);

And here is an example of what I am getting when I set dmp_sample_rate=50.

Time difference = 0.000546
Real Period = 0.019985
Time difference = 0.000652
Real Period = 0.020011
Time difference = 0.000842
Real Period = 0.019995
                                          ...... And goes up .....
Time difference = 0.004045
Real Period = 0.019998
Time difference = 0.004392
Real Period = 0.020040
Time difference = 0.004474
Real Period = 0.019898
                                        ..... And goes up.....
Time difference = 0.007977
Real Period = 0.019994
Time difference = 0.008268
Real Period = 0.020042
Time difference = 0.008476
Real Period = 0.019996
                                      ..... And goes up.....
Time difference = 0.011333
Real Period = 0.019978
Time difference = 0.011370
Real Period = 0.020002
Time difference = 0.011478
Real Period = 0.020093
                                     ..... All the way to.....
Time difference = 0.019607
Real Period = 0.020005
Time difference = 0.019588
Real Period = 0.019974
Time difference = 0.019687
Real Period = 0.020026

So as you can see, at some points the imu is giving me measures from 10 ms ago, at others from 19 ms ago, at other from 1 ms ago... so... this is too much variation, at least for my control strategy.

I post this for anyone out there implementing some discrete time control laws which relay heavily on a relatively precise/constant sampling time (which indeed means SAMPLING the sensor at that sampling time, not just running the code at that sampling time).

I also verified that my controller was MUCH MUCH MUCH more stable when I read (one-shot) without the dmp while flying acro mode.

Regards,
Oscar

msardonini, 2017-09-21:
I have also seen these non-deterministic reading times with this IMU/software. After some debugging, I'm fairly certain that it's the the fault of the IMU not having data ready in time (comments in rc_mpu9250.c also suggest this). 

Thanks for the insight that performance is improved by turning off the dmp feature, what fusion algorithm are you using in place of the dmp?

I've been recently testing how the IMU performs without the dmp, and I've found that if I sample at 100 Hz or greater, the IMU will occasionally 'hiccup' and the read calls will block the program for precisely 4.3 seconds, enough time for complete failure in my application (quad-copter program). Have you seen this? From what I can tell, sampling at 50 Hz has been working fine.

Mike

OscarJGV26, 2017-09-21:
Hi @msardonini 

Thank you for your response and your tips about sampling at 100Hz. I am also working in a quadcopter for now and yes, 4.3 means total catastrophe. XD

I am currently sampling at 50Hz and fusing the data for attitude estimation with the codes that I give in this repository (https://github.com/OscarJGV26/IMU_EKF). 

I replicated the algorithm from the paper (Double Stage Kalman Filter for IMU.pdf) that I also share in the repository and then converted to C code using Matlab C Coder. Renamed it to .cpp afterwards just for having all my files in cpp. If you wan't to quickly check the Matlab code with the paper, go to page 4, bottom/right section and you will see the exact steps replicated. I added a normalization of the quaternion at the end and normalization of accelerometer/magnetometer readings. Not sure if I should have though. I am still testing it... Works fine with Cov_info={0.00001,0.1,1}

If you want to use it just just create 
float Pcov[16];
float quat[4];
float Cov_info={0.00001,0.1,1}
float omega[3];
float accel[3];
float mag[3];
float dt=0.02;             //To be measured outside. You can also leave it as a constant
int ini=1;    //Just the first time you run it.
int use_mag=1;         //Use mag (1=Yes)

The call the function with,
IMU_EKF(Pcov,quat,Cov_info,omega,accel,mag,dt,ini,use_mag)

Note: Only initialize once at start up or if you want to reset the covariance/quaternion.

Feel free to try it out and if you like it please leave me a comment! :D

OscarJGV26, 2017-09-21:
Also watch out for the timing of the barometer reading. If you call it at 50Hz it will sometimes read and other times not, moving your code behaviour slightly (+/-400 microseconds). 

If you think is appropriate, consider reading it after you have made and written the control decision u_k to have deterministic/predictable timing in your decision variable going into the system.... 

Its less than 4% variation from the sampling time at 50Hz but the more stable the better! :D 

You can probably just use it in the next sampling time anyways... It's not like it is that precise/reliable/important. I mean its just for the correction step and it probably has lag anyways so...

StrawsonDesign, 2018-04-13:
some of the variation came from reading the magnetometer on some samples but not others, and some packets being dropped with bad data. By default now the magnetometer is read after the dmp for more consistent interrupt samples but can be put back to how it was in the config struct. I'm seeing very consistent results on my end now, I hope you see the same. Also I put in a workaround for the DMP sending interrupts at 200hz despite data only being ready at the requested rate. This really cut down on CPU use and dropped packets.

closed 2018-04-13 by StrawsonDesign