beagleboard/librobotcontrol #79: Magnetometer not active in DMP mode on Beaglebone Blue

logos-electromechanical, 2017-10-11:
I calibrated the magnetometer and confirmed that it is working as expected using rc_test_imu. 

However, when I run:
`rc_test_dmp -c -m -t`

The raw compass output reads about 0, 180, or -180. It changes seemingly at random as the BBBlue is rotated around any of the three axes. The filtered heading changes as expected when I rotate the BBBlue, but rapidly converges back to the raw compass reading (as expected). I get identical results with the magnetometer turned off, so I suspect that the issue is that the magnetometer is not being activated in DMP mode.

StrawsonDesign, 2018-04-13:
fixed in v0.4.0

closed 2018-04-13 by StrawsonDesign

logos-electromechanical, 2018-06-12:
Just noticed this, thanks for the fix! I'll check it out next time I'm working on the project I'm using this for.