beagleboard/librobotcontrol #80: rc_balance works fine but how do i open/run balance.py?

norstokke, 2017-10-23:
as per the title i can open and run rc_balance, thank you. I am not able to run balance.py that would allow me and my grandson to us the arrow keys for direction. 

thanks

wade

rahulsharma11, 2017-10-24:
Hi, Are you using sudo python3 rc_balance.py?

norstokke, 2017-10-24:
rahulsharma, thank you for the reply. sudo python3 rc_balance.py does not seem to work either.

rahulsharma11, 2017-10-24:
Can you please tell what exact error you are facing?

norstokke, 2017-10-24:
in cloud9 i get the following error

Your code is running at http://localhost.
Important: use os.getenv(PORT, 8080) as the port and os.getenv(IP, 0.0.0.0) as the host in your scripts!

Traceback (most recent call last):
  File "/var/lib/cloud9/examples/robot/python/button.py", line 14, in <module>
    GPIO.setup(button, GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

rahulsharma11, 2017-10-26:
I haven't use cloud9 for that. I have followed "https://github.com/mcdeoliveira/rcpy" to get that done. It's working.

closed 2018-04-18 by StrawsonDesign