beagleboard/librobotcontrol #84: installing from package on Debian 9.2 gives error

ycdtosa, 2017-11-20:
i have build the package from master with
make package.

then i install with
<pre>
sudo dpkg -i roboticscape_0.3.4_armhf.deb 
(Reading database ... 57047 files and directories currently installed.)
[...]
**********************************************
Configuring Device Tree
/usr/bin/configure_robotics_dt.sh: line 13: warning: command substitution: ignored null byte in input
ERROR: This is not Debian Jessie.
Flash the latest Jessie image to your BBB
or use the Wheezy branch of this installer.
**********************************************
 
Robotics Cape package installation complete.
Please reboot now.
 
**********************************************
</pre>

That error is triggered by this code:
<pre>
# make sure the release is really jessie
if ! grep -q "8." /etc/debian_version ; then
	echo "ERROR: This is not Debian Jessie."
	echo "Flash the latest Jessie image to your BBB"
	echo "or use the Wheezy branch of this installer."
	exit 1
fi
</pre>

which i tweaked to suit my needs ( just let it be in debian "9." )
and the library seems to work just fine afterwards.

my guess is that this library does work in both Debian 8.x and Debian 9.x

am i correct?

RobertCNelson, 2017-11-20:
The /etc/debian_version check can be nuked.
Regards,

StrawsonDesign, 2018-04-13:
fixed with cd7e1f9c8bdb4b1579dcc3dcbba5349b4166dea1

closed 2018-04-13 by StrawsonDesign