beagleboard/librobotcontrol #91: roboticscape device tree breaks usb network autoconfiguration in Linux

pgaskell, 2018-01-11:
When connecting to a beaglebone black or green after loading the robotics cape device tree the usb network auto-configuration (both rndis and cdc-ether) is broken, and requires a static IP to be set on the host computer.  It would be nice if this wasn't required, and the network interfaces automatically were configured.

ycdtosa, 2018-01-30:
I may have run into the same issue.
Do you have more info about this?

Thalaivar, 2018-01-31:
Is there anyway to compile programs using robotics cape on an Ubuntu pc?

StrawsonDesign, 2018-04-18:
I observed this a long time ago but sadly have no idea why since I only enable hardware (eeqep, pwm, etc) in that device tree. Any help would be appreciated!

Thalaivar, yes, and I do have it installed on my computers. I added a 'make native' target to the library makefile so it will use the native compiler instead of building specifically for ARM. The examples always compile natively anyway. Try this:

cd Robotics_Cape_Installer/library/
make native && sudo make install
cd ../examples
make && sudo make install

Obviously the device tree and PRU code are worthless on a PC. But things like the math/timing/start_stop subsystems work, as will standard driver things like UART, I2C, GPIO, SPI

StrawsonDesign, 2018-04-19:
@RobertCNelson I put up a pull request to fix this on your boot-scripts repo here https://github.com/RobertCNelson/boot-scripts/pull/82

I copied the contents of am335x-boneblack.dts verbatim into am335x-boneblack-roboticscape.dts and DHCP goes back to normal. However changing ONLY the model name from "TI AM335x BeagleBone Black" to "TI AM335x BeagleBone Black RoboticsCape" breaks DHCP again so I knew to look for a startup script switch case and the above pull request seems to do the trick.

closed 2018-04-19 by StrawsonDesign