beagleboard/librobotcontrol #92: roboticscape does not support Stretch

appliedrd, 2018-01-20:
beaglebone blue Rev A2
using https://debian.beagleboard.org/images/bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

sudo apt-get update
sudo apt-get upgrade roboticscape
sudo configure_robotics_dt.sh
/usr/bin/configure_robotics_dt.sh: line 13: warning: command substitution: ignored null byte in input
ERROR: This is not Debian Jessie.
Flash the latest Jessie image to your BBB
or use the Wheezy branch of this installer.

appliedrd, 2018-01-20:
just edit install.sh and change 8 to 9

ycdtosa, 2018-01-29:
see #84 which has a PR waiting

ycdtosa, 2018-02-11:
thi should be closed as duplicate of #84

StrawsonDesign, 2018-04-05:
duplicate of #84