beagleboard/librobotcontrol #96: DSM connection error

Sriherams, 2018-04-01:
Hi All,

I am trying to connect spektrum dsm8 receiver. After binding the receiver,I  ran "rc_test_dsm".

Here is the result:

WARNING: too many DSM channels detected, trying again
WARNING: too many DSM channels detected, trying again
WARNING: too many DSM channels detected, trying again
.................

Appreciate your help!

Thanks

StrawsonDesign, 2018-04-04:
which radio are you using this with?

Sriherams, 2018-04-04:
I am using Spektrum DX8 8-Channel Gen 2 Transmitter.

StrawsonDesign, 2018-04-18:
I got my hands on a DX8 and with an official DSMX receiver I can't replicate this, it works as expected. Please try with the current master branch v0.4.0 as I just pushed up some performance enhancements in the DSM lib which may have inadvertently fixed your issue. Closing for now since I can't replicate it with the same hardware.

closed 2018-04-18 by StrawsonDesign

StrawsonDesign, 2018-04-21:
Sriherams, I managed to recreate your problem but only with a HobbyKing Orange RX receiver which is great for spitting out useless packets. I improved the error checking and it seems to work with this new RX110 receiver I got which should hopefully fix your issue too. 0bb1c0c4bb6be5caaccaaef91a320a8c45c47284