beagleboard/librobotcontrol #97: BeagleBone Blue encoder 4

DocVaughan, 2018-04-10:
I'm working with your library on a BeagleBone Blue for a system that has 4 encoders. However, the 4th encoder reading doesn't seem to be working properly; it simply toggles between 0 and 1 as the encoder turns. I have checked that the encoder is working by using the other ports. They all work. I have also tried a second BeagleBone Blue and got the same result.

jadonk, 2018-04-11:
Which version of system software are you using? This is probably a temporary regression on our part and soon to be fixed in an upcoming release.

StrawsonDesign, 2018-04-12:
the PRU remoteproc interface changed a bit in recent kernels. V0.4.0 supports the new interface, please give this new deb package a go. 

https://github.com/StrawsonDesign/Robotics_Cape_Installer/releases/tag/v0.4.0

Just a heads up, the PRU encoder test program is now separate and called rc_test_encoder_pru to keep that subsystem separate and prevent future issues/changes from effecting the normal eqep subsystem and test program.

DocVaughan, 2018-04-13:
@RobertCNelson - I didn't see you message before installing the 0.4.0. The output after that install is:

```
git:/opt/scripts/:[d36fe9a7be9ebfc872b10a470e904ab4c61c4516]
eeprom:[A335BNLTBLA21712EL002069]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.03-rc2-00002-g11d4fd]
kernel:[4.4.91-ti-r133]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180408.0-0rcnee0~stretch+20180409]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
```

@StrawsonDesign - Perhaps I missed a step in the `.deb` package install, but the `rc_` commands were not recognized after installation via that method.

So, I installed from source. The `rc_` commands are recognized. Running `rc_test_encoder_pru` gives (with the same output if run with `sudo`):

```
$ rc_test_encoders_pru

ERROR in rc_pru_stop opening remoteproc driver: No such file or directory
PRU probably not enabled in device tree
ERROR in rc_encoder_pru_init, failed to start PRU0
ERROR: failed to run rc_encoder_pru_init
```

Other `rc_test_` commands seem to all be working fine, including `rc_test_encoders_eqep`.

RobertCNelson, 2018-04-13:
@DocVaughan 0.4.0 needs v4.14.x-ti pru..

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.03-rc2-00002-g11d4fd]
```

First, nuke the eMMC bootloader via;
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

2nd: update the bb-cape-overlays, this contains the overlays:
```
sudo apt update ; sudo apt install --only-upgrade bb-cape-overlays
```

3rd: tweak /boot/uEnv.txt
```
from:
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo

to:
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```

4th: update /opt/scripts/
```
cd /opt/scripts/
git pull
```

5th: update kernel
```
sudo /opt/scripts/tools/update_kernel.sh --ti-channel --lts-4_14
```

6th: update bootloader
```
sudo /opt/scripts/tools/developers/update_bootloader.sh
```

Regards,

StrawsonDesign, 2018-04-20:
4.9 also works, I've been developing recently on the latest 'official' stable image from beagleboard.org which is using 4.9.82-ti-r102

Embarrassingly I went to confirm this works myself and found a quick typo causing the encoder pru binary to load but always return a position of 0. fixed now in master. Pushing a new deb package momentarily.

Closing issue for now since it seems to be working with the right kernel. We can keep discussing this here if you still need help getting it going.

closed 2018-04-20 by StrawsonDesign

unl0ck, 2022-02-17:
it is still broken encoder 4 using kernel 4.19 newest kernel ti and librobotcontrol