beagleboard/librobotcontrol #101: Can't get rc_test_escs to work

jeholliday, 2018-04-28:
I've been testing 0.4.2/0.4.1 to see if I can get it to work before I move all my software over. I'm getting stuck when trying to run the rc_test_escs example. I can successfully run the other examples, but this specific one, is giving me errors no matter what input I give it.

Here is the error:
```
ERROR in rc_servo_send_esc_pulse_normalized, normalized input must be between -0.1 & 1.0
ERROR in rc_servo_send_esc_pulse_normalized, normalized input must be between -0.1 & 1.0
ERROR in rc_servo_send_esc_pulse_normalized, normalized input must be between -0.1 & 1.0
ERROR in rc_servo_send_esc_pulse_normalized, normalized input must be between -0.1 & 1.0
```

I've tried both 0.4.2 and 0.4.1 and had the same issue in both versions.
Here is my  version output:
```
git:/opt/scripts/:[5935e83ecaedfd4fd88a4b870ffe5af33ded0617]
eeprom:[A335BNLTBLA21712EL002543]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.9.88-ti-r111]
nodejs:[v6.14.1]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180424.0-0rcnee0~stretch+20180424]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.458682] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```

I'm not sure if this is an error in the example, the library, or my setup, but I would appreciate any assistance.

StrawsonDesign, 2018-04-29:
Good catch, it was a simple error when comparing a float to a double. d120db98d4077309e6003d1ebe16a5fced47b1fe should fix it. I don't have an ESC on hand to confirm but the program runs for me so please let me know and report back. The actual bugfix is only in the library so just git pull && cd library && make && sudo make install

jeholliday, 2018-05-03:
I can confirm that the changes to the library work, but I don't think that you meant to introduce a return statement at line 245 in rc_test_escs.c. Additionally I would appreciate a flag to disable the calibration wake up because my esc doesn't have a calibration mode. Instead it drives my esc a full reverse for 3 seconds.

StrawsonDesign, 2018-05-06:
good idea, added in 8a98e7db876048d2f080c739d0317ed294ddecef although right now the latest 4.9 and 4.14 kernels have something wrong with PRUSS which breaks servos

closed 2018-05-08 by StrawsonDesign