beagleboard/librobotcontrol #102: Using this library for cross compilation

gudozhni, 2018-05-01:
Hello,

I'm trying to setup the environment for developing for BeagleBone on a Debian machine.
Do you have any advice on how to install roboticscape lib there? Just copying the source files didn't help :(

Thanks in advance for any hint!

cheers,
Olex

jeholliday, 2018-05-04:
With 0.3.4, I was able to compile just the library folder and generate and install a .so file and headers. This allowed me check for compilation on an x86 machine.
With the update to 0.4, I can no longer build the library folder. I'm stuck on trying to resolve "linux/gpio.h". My research lead me to actually finding it in /usr/src/...,but I'm unsure how to properly setup the environment, so that building properly resolves this location.

StrawsonDesign, 2018-05-05:
interesting, I compile and run this lib on my debian and ubuntu machines just fine by running make && sudo make install inside the library directory. "/usr/include/linux/gpio.h " exists on both my debian stretch and ubuntu 18.04 machines. Can you be more specific about the environment you are compiling on?

StrawsonDesign, 2018-06-22:
I added an install.sh script to the root directory that asks if you want to only install the library and examples which is all you should want for non-beaglebone platforms.

closed 2018-06-22 by StrawsonDesign