beagleboard/librobotcontrol #104: ADC is broken in 0.4.2

ningen, 2018-05-03:
I am trying to read battery voltage using rc_adc_batt() and before that I call rc_adc_init() which fails with error:
```
ERROR in rc_adc_init, failed to open iio adc interface
: No such file or directory
Perhaps kernel or device tree is too old
```

Running rc_test_drivers shows:
```
Kernel: 4.4.54-ti-r93
BeagleBoard.org Debian Image 2017-03-19
Debian: 9.4

ERROR:  gpio 0 driver not loaded
ERROR:  gpio 1 driver not loaded
ERROR:  gpio 2 driver not loaded
ERROR:  gpio 3 driver not loaded
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
ERROR:  pru-rproc driver not loaded
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
ERROR:  ADC iio driver not loaded
```

How can I load iio driver?

jeholliday, 2018-05-04:
#97 references that v4.0 requires at least kernel version 4.9. I've also found it helpful to update the bootloader.

Altair33, 2018-05-04:
I have the same problem with a beaglebone blue using the latest image from beaglebone after upgrading to 0.4.2

`debian@beaglebone:~$ rc_test_adc `
`ERROR in rc_adc_init, failed to open iio adc interface `
`: No such file or directory`
`Perhaps kernel or device tree is too old`
`ERROR: failed to run rc_init_adc()`

And only the adc driver is missing: 

`debian@beaglebone:~$ rc_test_drivers `
`Kernel: 4.9.82-ti-r102`
`BeagleBoard.org Debian Image 2018-03-05`
`Debian: 9.4`
`PASSED: gpio 0`
`PASSED: gpio 1`
`PASSED: gpio 2`
`PASSED: gpio 3`
`PASSED: pwm1`
`PASSED: pwm2`
`PASSED: eqep0`
`PASSED: eqep1`
`PASSED: eqep2`
`PASSED: pru-rproc`
`PASSED: uart1`
`PASSED: uart2`
`PASSED: uart4`
`PASSED: uart5`
`PASSED: i2c1`
`PASSED: i2c2`
`PASSED: spi`
`PASSED: LED`
`ERROR:  ADC iio driver not loaded`

ningen, 2018-05-04:
Thanks @jeholliday. I followed upgrade instructions in pull request you mentioned and that fixed adc problem but now I have mpu problem:

```
$ rc_test_mpu 
ERROR: in rc_i2c_init, ioctl slave address change failed
failed to initialize i2c bus
rc_mpu_initialize_failed

$ rc_test_drivers

Kernel: 4.14.37-ti-r46
BeagleBoard.org Debian Image 2017-03-19
Debian: 9.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
ERROR:  uart4 driver not loaded
PASSED: uart5
ERROR:  i2c1 driver not loaded
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio
```

Updating kernel and bootloader didn't solve adc problem, I had to add:
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```
and also uncomment:
```
enable_uboot_overlays=1
```
in `/boot/uEnv.txt`. But now uart4 and i2c1 are broken. Do I need more magic lines in `/boot/uEnv.txt`?

ningen, 2018-05-05:
Adding:

```
uboot_overlay_addr0=/lib/firmware/BB-I2C1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-UART4-00A0.dtbo
```

to `/boot/uEnv.txt` made driver checks to pass but MPU is still not working:

```
debian@beaglebone:~$ rc_test_drivers 

Kernel: 4.14.37-ti-r46
BeagleBoard.org Debian Image 2017-03-19
Debian: 9.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

debian@beaglebone:~$ rc_test_mpu 
ERROR: in rc_i2c_init, ioctl slave address change failed
failed to initialize i2c bus
rc_mpu_initialize_failed
```

StrawsonDesign, 2018-05-05:
yup, this effects me too, but it's a device-tree problem that can't be fixed in this repo. I've submitted a pull request with a fix for 4.14 which sort this out.

closed 2018-05-08 by StrawsonDesign