beagleboard/librobotcontrol #107: Build Failure On Master: 0.4.3-git20180508.0 git : 3d35ecbd12b7cae475cc07085cea78dea3cc53a9

RobertCNelson, 2018-05-08:
Build log:

```
make[1]: Entering directory '/<<PKGBUILDDIR>>'
CC	src/main_pru0.c
CC	src/pru0-encoder.asm
LD	bin/main_pru0.object bin/pru0-encoder.object
CC	src/main_pru1.c
CC	src/pru1-servo.asm
LD	bin/main_pru1.object bin/pru1-servo.object
Generated: bin/am335x-pru0-rc-encoder-fw bin/am335x-pru1-rc-servo-fw
made: build/pthread.o
made: build/mavlink_udp.o
made: build/start_stop.o
made: build/version.o
made: build/time.o
made: build/led.o
made: build/bmp/bmp.o
made: build/math/matrix.o
made: build/math/algebra.o
made: build/math/kalman.o
made: build/math/algebra_common.o
made: build/math/quaternion.o
made: build/math/other.o
made: build/math/filter.o
made: build/math/polynomial.o
made: build/math/vector.o
made: build/math/ring_buffer.o
made: build/dsm.o
made: build/io/uart.o
made: build/io/adc.o
made: build/io/spi.o
made: build/io/i2c.o
made: build/io/encoder_eqep.o
made: build/io/gpio.o
made: build/io/pwm.o
made: build/mpu/mpu.o
made: build/pru/encoder_pru.o
made: build/pru/pru.o
made: build/pru/servo.o
made: build/model.o
made: build/cpu.o
made: build/motor.o
made: build/deprecated.o
made: build/button.o
made: build/pinmux.o
Done making lib/libroboticscape.so.0
/usr/bin/ld: cannot find -lroboticscape
collect2: error: ld returned 1 exit status
Makefile:62: recipe for target 'bin/rc_test_algebra' failed
make[2]: *** [bin/rc_test_algebra] Error 1
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/<<PKGBUILDDIR>>'
dh_auto_build: make -j1 returned exit code 2
debian/rules:3: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
--------------------------------------------------------------------------------
Build finished at 2018-05-08T14:00:35Z

Finished
--------

+------------------------------------------------------------------------------+
| Cleanup                                                                      |
+------------------------------------------------------------------------------+

Purging /<<BUILDDIR>>
Not cleaning session: cloned chroot in use
E: Build failure (dpkg-buildpackage died)

+------------------------------------------------------------------------------+
| Summary                                                                      |
+------------------------------------------------------------------------------+

Build Architecture: armhf
Build Type: full
Build-Space: 7960
Build-Time: 35
Distribution: stretch
Fail-Stage: build
Host Architecture: armhf
Install-Time: 146
Job: roboticscape_0.4.3-git20180508.0-0rcnee0~stretch+20180508.dsc
Machine Architecture: armhf
Package: roboticscape
Package-Time: 223
Source-Version: 0.4.3-git20180508.0-0rcnee0~stretch+20180508
Space: 7960
Status: attempted
Version: 0.4.3-git20180508.0-0rcnee0~stretch+20180508
--------------------------------------------------------------------------------
Finished at 2018-05-08T14:00:35Z
Build needed 00:03:43, 7960k disk space
E: Build failure (dpkg-buildpackage died)

```

RobertCNelson, 2018-05-14:
Closing, Fixed on master

closed 2018-05-14 by RobertCNelson

StrawsonDesign, 2018-05-14:
I love when issues close themselves! Yes, in 0.4.3 I took a stab at quietening the lintian warnings about not using a version number in the .so but it turned out to be more trouble than it was worth since LD didn't seem to make an unversioned symlink like lintian claimed it would.