beagleboard/librobotcontrol #108: pru-rproc driver not loading?

pgaskell, 2018-05-18:
I am testing the latest roboticscape 0.44, with the latest Stretch iot image running kernel 4.9.88-ti-rt-r111.  It appears as the pru-rproc driver is loaded...

debian@beaglebone:~$ lsmod | grep pru
pru_rproc              14893  0
pruss_intc              9095  1 pru_rproc
pruss                  11017  1 pru_rproc

but when I run rc_test_drivers, I get:
...
ERROR:  pru-rproc driver not loaded
...

When I run any of the rc_test programs that require PRU firmware (rc_test_servos, rc_test_encoders_pru) I get a Bus Error:

Program received signal SIGBUS, Bus error.
0xb6f10998 in rc_encoder_pru_init () at src/pru/encoder_pru.c:34
34		shared_mem_32bit_ptr[ENCODER_MEM_OFFSET]=42;

Any ideas?

StrawsonDesign, 2018-05-21:
the development branch of 4.9 is still a bit iffy with prurproc. Try the may 10th image which is running a 4.14 kernel that seems to have PRU working. https://rcn-ee.net/rootfs/bb.org/testing/2018-05-10/stretch-iot/

You may also have an old bootloader that's also needed for the PRU to work so best to flash the eMMC with the BBB-blank image (for black) or BBBL-blank image (for blue).

pgaskell, 2018-05-21:
I load the may 10th image and run rc_test_drivers and run rc_test_drivers and get:

debian@beaglebone:~$ rc_test_drivers 

Kernel: 4.14.39-ti-r47
BeagleBoard.org Debian Image 2018-05-10
Debian: 9.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
ERROR:  uart5 driver not loaded
PASSED: i2c1
PASSED: i2c2
PASSED: spi
ERROR:  LED driver not loaded
PASSED: ADC iio

then I run configure_robotics_dt, reboot and rerun rc_test_drivers and I get:

debian@beaglebone:~$ rc_test_drivers 

Kernel: 4.14.39-ti-r47
BeagleBoard.org Debian Image 2018-05-10
Debian: 9.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
ERROR:  pru-rproc driver not loaded
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

There must be something wrong with the robotics cape device tree?

pgaskell, 2018-05-23:
I added  to /boot/uEnv.txt the following lines to enable uboot overlays and the overlay for the PRU and can now get all drivers to pass and all 4 encoders reading.

...
enable_uboot_overlays=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo
...

using Stretch image: bone-debian-9.4-iot-armhf-2018-04-22-4gb.img
kernel: 4.9.88-ti-rt-r111
bootloader: U-Boot 2018.03-00002-gac9cce7c6a

ctmrocks2435, 2018-05-29:
I'm also having this issue, I have tried @pgaskell's solution and it did not work for me. I have also tried with the following image configuration:

Kernel: 4.14.40-ti-r50
BeagleBoard.org Debian Image 2017-10-10
Debian: 9.4

I can not get the pru to load.
I'll keep trying alternative configurations unless someone has a better idea.

RobertCNelson, 2018-05-29:
@ctmrocks2435 u-boot's probably too old..  run the version.sh script so we can verify:

```
cd /opt/scripts/tools/
/opt/scripts/tools$ git pull
/opt/scripts/tools$ sudo ./version.sh 
```

Regards,

ctmrocks2435, 2018-05-29:
The results of the "/opt/scripts/tools$ sudo ./version.sh " is below, I still have uEnv.txt modified per  pgaskell's suggestion, I'm not sure if that will screw up this script or not. Tonight I plan to try an eMMC flash with one of the development images from the repository you posed above. It sounds like the "Blank" images are setup already to overwrite eMMC?
"BBB-blank-debian-9.4-iot-armhf-2018-05-27-4gb.img.xz"

```debian@beaglebone:/opt/scripts/tools$ sudo ./version.sh
git:/opt/scripts/:[92d9c6bcdf437a3a1b3933e9a415496398d9b16c]
eeprom:[A335BNLT00C02917BBBK028D]
model:[TI_AM335x_BeagleBone_Black_RoboticsCape]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]:[location: dd MBR]
kernel:[4.14.40-ti-r50]
nodejs:[v6.14.2]
device-tree-override:[dtb=am335x-boneblack-roboticscape.dtb]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo]
pkg:[bb-cape-overlays]:[4.4.20180424.0-0rcnee0~stretch+20180424]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M]
dmesg | grep pinctrl-single
[    1.057100] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper```

ctmrocks2435, 2018-05-30:
****Update on more troubleshooting:****
I downloaded the BBB-Blank image from: [here](https://rcn-ee.net/rootfs/bb.org/testing/2018-05-27/stretch-iot/)
Then I burned it to an SD card, and flashed it to the eMMC. I guess this also updates some "firmware" like u-boot etc.
Upon startup I ran `sudo rc_test_drivers`
Below is the output, no PWM or LEDs
```
Kernel: 4.14.40-ti-r50
BeagleBoard.org Debian Image 2018-05-27
Debian: 9.4

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
ERROR:  uart5 driver not loaded
PASSED: i2c1
PASSED: i2c2
PASSED: spi
ERROR:  LED driver not loaded
PASSED: ADC iio
```
I figured maybe the overlay is not correct, so I ran `sudo configure_robotics_dt`
```
Using am335x-boneblack-roboticscape.dtb
making backup copy of /boot/uEnv.txt
Using previously listed kernel 4.14.40-ti-r50
Robotics Cape Device Tree Configured and Installed
```
Then I rebooted only to be greeted by the same unsatisfying issue:
```PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
ERROR:  pru-rproc driver not loaded
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio
```
PWM still does not work with a "bus error"
Any more advice? Should I try another older image maybe?

Here is my new `sudo ./version` command
```
git:/opt/scripts/:[92d9c6bcdf437a3a1b3933e9a415496398d9b16c]
eeprom:[A335BNLT00C02917BBBK028D]
model:[TI_AM335x_BeagleBone_Black_RoboticsCape]
dogtag:[BeagleBoard.org Debian Image 2018-05-27]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.40-ti-r50]
nodejs:[v6.14.2]
device-tree-override:[dtb=am335x-boneblack-roboticscape.dtb]
pkg:[bb-cape-overlays]:[4.4.20180424.0-0rcnee0~stretch+20180424]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M]
dmesg | grep pinctrl-single
[    1.050815] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```

StrawsonDesign, 2018-05-30:
@RobertCNelson  Robert, I don't think this is U-Boot related. The device tree I have (and pushed to you) for the Robotics Cape was working in the past with 4.14 but it seems the status="okay" one-line device tree entry compiles and loads but isn't sufficient anymore. I updated my local copy of the device tree fragment in c9412edac68a5b68bf23c820e38baf75292acc78 to work in 4.14. I'm rather confused why the PRU works with the Blue without needing this entry at all, perhaps it's included with cape-universal or something else in there. 

I was going to send PR for this but I see your linux repo is being reorganized right now. Please let me know where to send or feel free to just copy the diff from the above commit. Also, I can't for the life of me get the PRU up on the latest 4.9 kernel. It's not the end of the world if the plan is to keep stretch at 4.14, I was thought it would be nice to keep the 4.9 option working. I was hoping to get it working to include with the 4.14 PR.

pgaskell, 2018-05-30:
Thank you @StrawsonDesign I  will probably move to the 4.14 kernel going forward.

closed 2018-05-30 by pgaskell