beagleboard/librobotcontrol #117: Position estimation with Kalman Filter example

rahulsharma11, 2018-07-01:
Hi,
I have been using roboticscape from last year. I also have successfully  integrated roboticscape library with ardurover project. It's wonderful library.
The question is for latest upgrad which is having kalman filter example. 
Does it give pos, vel of BBBlue by imu integration? I am seeing a considerable drift while keeping my BBBlue steady. If not then what it gives? I have seen the comment section on the top of the code(1 kg mass back-forth motion), did it meant for that purpose only ? Or something else is required to be configuration.
Will appreciate the elaboration on this example.
Thanks.

StrawsonDesign, 2018-07-01:
Hi Rahul,

I'm glad you like the library! rc_test_kalman is just part of the test suite to confirm the math works, it doesn't communicate with sensors. Try poking at the rc_altitude example, it actually estimates altitude, vertical velocity, and accelerometer bias. Please do make sure the gyroscope and accelerometer are calibrate first or it will be way off :)

Expanding rc_altitude to also estimate horizontal velocity with some high-pass rolloff wouldn't be too difficult since the example already takes care of orientation estimation and rotation of the accelerometer data from inertial to global reference frame. 2 more states would just need to be added to the system for the additional velocities. Adding estimation for the additional biasses would be better done by expanding from linear to EKF.

Let me know if this is something you are interested in playing with, I'd happily merge it into the library for others to use and learn from.

Thanks,
James

rahulsharma11, 2018-07-01:
Hi,
Thanks for the valuable and quick reply. That gave me clarity as well.
The thing i was doing from few months for one of my project is to get the position estimation with imu integration. Then i will fuse that data with encoder odometry to get the bot navigation more precisely. I know the drift concept while accelerometer integration. And i saw the upgraded version of roboticscape, didn't wait to test that how much it can help me.:-).
Rather than going with whole ardurover navigation project with BBBlue, i was thinking if BBBlue itself can provide the position estimation(2D plan) as a standalone example project. That will be very cool as well.
That's what i am planning to do.

rahulsharma11, 2018-07-01:
What i got from altitude example is imu integrated with baro to get z-axis estimation. 
Need to dig around x-y plan.

StrawsonDesign, 2018-07-01:
sounds like a neat project, and exactly the sort of thing I was hoping people would use the BB Blue for. Please stay in touch, I'd love to see how your project turns out :)

closed 2018-07-01 by StrawsonDesign