beagleboard/librobotcontrol #121: libroboticontrol: cannot find -lroboticscape

RobertCNelson, 2018-07-06:
```bash
 git clone https://github.com/mcdeoliveira/rcpy
 cd rcpy
 /usr/bin/python3 setup.py install
```

```bash
13:28:36 Cloning into '/opt/source/rcpy'...
13:28:39 running install
13:28:39 running bdist_egg
13:28:39 running egg_info
13:28:39 creating rcpy.egg-info
13:28:39 writing rcpy.egg-info/PKG-INFO
13:28:39 writing dependency_links to rcpy.egg-info/dependency_links.txt
13:28:39 writing top-level names to rcpy.egg-info/top_level.txt
13:28:39 writing manifest file 'rcpy.egg-info/SOURCES.txt'
13:28:39 reading manifest file 'rcpy.egg-info/SOURCES.txt'
13:28:39 reading manifest template 'MANIFEST.in'
13:28:39 writing manifest file 'rcpy.egg-info/SOURCES.txt'
13:28:39 installing library code to build/bdist.linux-armv7l/egg
13:28:39 running install_lib
13:28:39 running build_py
13:28:39 creating build
13:28:39 creating build/lib.linux-armv7l-3.6
13:28:39 creating build/lib.linux-armv7l-3.6/rcpy
13:28:39 copying rcpy/clock.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:39 copying rcpy/mpu9250.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:39 copying rcpy/led.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/encoder.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/adc.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/motor.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/__init__.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/gpio.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/servo.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 copying rcpy/button.py -> build/lib.linux-armv7l-3.6/rcpy
13:28:40 creating build/lib.linux-armv7l-3.6/test
13:28:40 copying test/test_motor.py -> build/lib.linux-armv7l-3.6/test
13:28:40 copying test/test_rcpy.py -> build/lib.linux-armv7l-3.6/test
13:28:40 copying test/test_encoder.py -> build/lib.linux-armv7l-3.6/test
13:28:40 copying test/test_clock.py -> build/lib.linux-armv7l-3.6/test
13:28:40 copying test/__init__.py -> build/lib.linux-armv7l-3.6/test
13:28:40 copying test/test_mpu9250.py -> build/lib.linux-armv7l-3.6/test
13:28:40 copying test/test_led.py -> build/lib.linux-armv7l-3.6/test
13:28:40 running build_ext
13:28:40 building 'rcpy._rcpy' extension
13:28:40 creating build/temp.linux-armv7l-3.6
13:28:40 creating build/temp.linux-armv7l-3.6/src
13:28:40 arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c src/_rcpy.c -o build/temp.linux-armv7l-3.6/src/_rcpy.o
13:28:40 arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.6/src/_rcpy.o -lroboticscape -lrt -o build/lib.linux-armv7l-3.6/rcpy/_rcpy.cpython-36m-arm-linux-gnueabihf.so
13:28:40 /usr/bin/ld: cannot find -lroboticscape
13:28:40 collect2: error: ld returned 1 exit status
13:28:40 error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

@jadonk & @StrawsonDesign I don't remember if wanted to it to be 100% compabile, looks like we need more then just a symlink... "-lroboticscape"

Regards,

StrawsonDesign, 2018-07-07:
oops, been fiddling with documentation locally and never pushed that commit up

148ef0299bf76240807be216a7b17a7ce2a020b4 adds the symlink, gcc manages to work with that symlink on my end.

RobertCNelson, 2018-07-07:
@StrawsonDesign looks good! Builder passed!  Tomorrow's buster image will have librobotocontrol, i'll migrate bionic/stretch on monday..

Regards,

closed 2018-07-07 by RobertCNelson