beagleboard/librobotcontrol #123: rc_test_servos does not work?

jvon-challenges, 2018-07-11:
I am trying to run the servo test on a recently updated BeagleBone Blue and I see:

root@beaglebone:/var/lib/cloud9# rc_test_servos -c 1 -p 0.0                                                                                                                                                                              
ERROR in rc_adc_init, failed to open iio adc interface
: No such file or directory
Perhaps kernel or device tree is too old
ERROR: failed to run rc_adc_init()
root@beaglebone:/var/lib/cloud9#

RobertCNelson, 2018-07-11:
Please run:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

michaelmegliola, 2018-07-11:
root@beaglebone:/var/lib/cloud9# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLTBLA21712EL000782]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.03-rc2-00002-g11d4fd]:[location: dd MBR]
kernel:[4.9.82-ti-r102]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180628.0-0rcnee0~stretch+20180628]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.436183] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
root@beaglebone:/var/lib/cloud9#

RobertCNelson, 2018-07-11:
@michaelmegliola 2018-03-05 that was 'buggy'....

Either follow below, or just flash 2018-06-17 from: https://beagleboard.org/latest-images

To update 2018-03-05 to 2018-06-17 (well the important bits) use this simple 5 step program

Clear eMMC's bootloader, it's breaking things going forward
```bash
debian@beaglebone:~$ sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

Update microSD's bootloader
```bash
debian@beaglebone:~$ cd /opt/scripts/tools/developers/
debian@beaglebone:/opt/scripts/tools/developers$ git pull
Already up-to-date.
debian@beaglebone:/opt/scripts/tools/developers$ sudo ./update_bootloader.sh
```

Update bb-cape-overlays pkg:
```bash
debian@beaglebone:~$ sudo apt update ; sudo apt install --only-upgrade bb-cape-overlays
```

Update to new v4.14.x kernel:
```bash
debian@beaglebone:~$ sudo /opt/scripts/tools/update_kernel.sh --ti-channel  --lts-4_14
```
Enable proper pru overlay for v4.14.x, open /boot/uEnv.txt
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
```

Update to librobotcontrol pkg:
```bash
sudo apt update ; sudo apt install  --only-upgrade librobotcontrol
```

That'll sync you up with:

```bash
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[5d0cce48cbd05aafa0d1e7cc1583f5f9a57c199c]
eeprom:[A335BNLTBLA21722EL001993]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-07-08]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.53-ti-r62]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180628.0-0rcnee0~stretch+20180628]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[roboticscape]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.026024] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```

```bash
debian@beaglebone:~$ sudo rc_test_servos -c 1 -p 0.0
Turning On 6V Servo Power Rail

Sending only to channel 1.
Using rc_servo_send_pulse_normalized
Normalized Signal: 0.000000  Pulse Frequency: 50
```

Regards,

StrawsonDesign, 2018-07-11:
Robert, do you mind clarifying why/when it's necessary to wipe the bootloader. Is this done automatically by the emmc flasher?

RobertCNelson, 2018-07-11:
@StrawsonDesign yes, if he updates to the new version via the eMMC flasher then updating the bootloader isn't required.

closed 2018-07-11 by StrawsonDesign