beagleboard/librobotcontrol #124: link_to_startup_program cannot be found.

rthoma82, 2018-08-01:
**Describe the bug**
I am in the process of learning how to setup a beaglebone blue to be used with the EduMIP kit. I was successful in my first attempt following all the advice and instructions found online. I decided to wipe everything and start over to be sure that I could do it again and this time document my process. While trying to unlink the rc_balance program from the link_to_startup_program, I accidentally deleted the original rc_balance file. After several attempts to recover or replace it, I ultimately had to reflash the board. After doing so, rc_balance was back in place, but when I went to link it to /etc/roboticscape/link_to_startup_program I got an error saying that it was not found. Sure enough, the roboticscape directory is empty. 

In addition, I do believe that I am now running a different version of Debian after the reflash because it looks somewhat different. I don't know for sure what version I started on (8.9 is my best educated guess), but the current version is 9.4. Not sure if that would have anything to do with the missing file.

All I really want is to get the rc_balance program to run on boot again, but I am brand new to all of this and not sure what else to do since the only instructions I have found to make this happen, http://strawsondesign.com/#!edumip-software , do not work. Can you offer any help?

StrawsonDesign, 2018-08-01:
While you should never need to reflash the whole thing to recover one file, it's best that you are now up to date with a new image. Which image did you flash btw?

The reason for the missing folder is that the software package changed recently to be called "librobotcontrol" instead of roboticscape since it's no longer specific to the robotics cape. More detail about the file in the library documentation here: http://strawsondesign.com/docs/librobotcontrol/runonboot.html

Please join us in the Slack workspace to continue chatting further.: https://join.slack.com/t/librobotcontrol/shared_invite/enQtMzk0NjkxNjk0MTE0LTliNjFmMjZiZTQxOWZmNWI3NTJkN2UwOTdjNDk2OWQwMDgxNDEyMDNkMzBkODE5ODZiMWJkYmM1M2UwOTczNmE

As you are aware, documentation is critical for a good user experience. I've tried to write in as much detail as I can but what's obvious to me may not be for other users. Since you are trying to document your process I hope you will share and make suggestions to make the public docs better.

closed 2018-08-01 by StrawsonDesign

rthoma82, 2018-08-01:
I flashed 9.4. I did end up having to reflash because in my attempts to recover that one file, screwed a whole bunch of other things up. Anyway, I’ve looked over the link you sent and have tried a couple of things in it but so far without any success. I still can’t find the link_to_start_program even with the updated info. I will keep looking though. I also tried using the template to make a bootfile but I don’t know exactly what I’m missing.

Here’s what I’ve got so far:
Did this: cp -r /usr/share/robotcontrol/rc_project_template ~/new_project_name
cd new_project_name
No problems there. Then renamed the project name as instructed and changed the TARGET to match.
        mv rc_project_template.c new_project_name.c
        nano Makefile

After making those changes, I saved the Makefile and exited. This where things got confusing I know I need to link the file I want to start on boot to my project but did not know where the sudo make runonboot came into play which is listed in the making a program run on boot section.
I linked the rc_balance.c to rc_start_test.c which is what I named my new project. I used ln -s -f to link them since that was what it said in the original instructions on how to set it up. I tested this but it did not work. What am I missing?

Do I need to save the Makefile as something else?
When do I use the sudo make runonboot?

When I do get all of my process stuff gathered together, where do I share it?

Richard Thomas
Sr. Electronic Technician
Kent State University - Tuscarawas
(330) 308-7559

StrawsonDesign, 2018-08-01:
link_to_startup_program doesn't exist until you create it. That's what make runonboot does (or by making the link manually). Also, the link needs to point to the compiled executable, not a .c source code file. Also make sure you're looking in /etc/robotcontrol/ not /etc/roboticscape.

9.4 only indicates the minor release milestone of the base debian repositories and doesn't indicate the BeagleBoard.org image version which is the date listed in the file name. Beagleboard builds images several times a week and things move very rapidly. After flashing, you can check the image, library, and kernel version with:

rc_version
rc_test_drivers

Please join us on the slack workspace to keep discussing further so we can keep discussion separate from the bug tracker. https://join.slack.com/t/librobotcontrol/shared_invite/enQtMzk0NjkxNjk0MTE0LTliNjFmMjZiZTQxOWZmNWI3NTJkN2UwOTdjNDk2OWQwMDgxNDEyMDNkMzBkODE5ODZiMWJkYmM1M2UwOTczNmE