beagleboard/librobotcontrol #125: PWM initialization should use memory map for index

jadonk, 2018-08-09:
**Describe the bug**
librobotcontrol currently relies on indexes that aren't necessarily fixed across kernel versions.

See https://github.com/beagleboard/linux/issues/179 for related kernel entry.

Here's an alternative means for discovering the associated pwmchip entry for each hardware PWM.
```sh
debian@beaglebone:/var/lib/cloud9$ ls /sys/devices/platform/ocp/483*.epwmss/483*.pwm/pwm/pwmchip*
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1:
device  export  npwm  power  subsystem  uevent  unexport

/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip4:
device  export  npwm  power  subsystem  uevent  unexport

/sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip7:
device  export  npwm  power  subsystem  uevent  unexport
```

**To Reproduce**
1. Use the latest test build and kernel.
```sh
debian@beaglebone:/var/lib/cloud9$ rc_test_motors -s 0.3
ERROR in rc_pwm_init, can't open pwm unexport file for writing: No such file or directory
Probably kernel or BeagleBone image is too old
ERROR in rc_motor_init, failed to initialize pwm subsystem 1
```

**Expected behavior**
EduMIP motors should have turned in alternate directions.

**Platform information**
git:/opt/scripts/:[4a61888e8e27158bcbb0ca70531038d225a8d7af]
eeprom:[A335BNLTBLA21708EL000277]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-08-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.07-rc1-00002-g4d87f7]:[location: dd MBR]
kernel:[4.14.61-ti-r68]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.0-git20180726.0-0rcnee0~stretch+20180727]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.043124] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper

StrawsonDesign, 2018-08-09:
oh my, someone submitted a bug, and used the template!!!! Hallelujah, way to keep the standards high Jason!!!! :-D

I tried to be a little bit fancy and use globs so if the indexing changes from 0,2,4 to 0,3,7 or some other random sequence it should keep working. I just tried this with 4.9, 4.14.54 and the current 4.14.61. All seem to run with the pwmfix branch, but I'll try tomorrow with a motor to confirm before pulling into master and marking version 1.0.1 since this is a fairly important update.

My glob approach if anyone is interested: 4fa8abc5a0bf51bbca2a74825cddbb30deac7619

Also, who do we point fingers at in TI for this? I feel like this is the 4th time I've had to update the PWM interface to reflect non-backward-compatible changes in the PWM driver.

jadonk, 2018-08-09:
Not really a TI thing, it is more of a Linux thing.  Check the 'blame' log and maybe we can find the culprit.

https://github.com/torvalds/linux/blame/master/drivers/pwm/pwm-tiehrpwm.c 
https://github.com/beagleboard/linux/blame/4.14/drivers/pwm/pwm-tiehrpwm.c

Wanted to note rc_test_drivers needed to be updated separately, since it doesn't use the library to perform the PWM driver test.

StrawsonDesign, 2018-08-10:
mmm, thanks for the reminder. All sorted now and new minor version pushed out.

closed 2018-08-10 by StrawsonDesign