beagleboard/librobotcontrol #129: Installation / compatible kernel?

michaelmegliola, 2018-08-21:
I am trying to install version 1.0 on a BeagleBone Blue but cannot upgrade to the recommended version of debian; the Strawson web page points here:

The requested URL /rootfs/bb.org/testing/2018-07-15/stretch-iot/BBBL-blank-debian-9.5-iot-armhf-2018-07-15-4gb.img.xz was not found on this server.

Our BeagleBone Blue's are on kernel 4.14.49-ti-r54

That version will not access servos:

root@beaglebone:/var/lib/cloud9# rc_test_servos -c 1 -p 0.0
ERROR in rc_adc_init, failed to open iio adc interface
: No such file or directory
Perhaps kernel or device tree is too old
ERROR: failed to run rc_adc_init()

Where is the best stable combination of debian & librobotcontrol?

RobertCNelson, 2018-08-21:
@michaelmegliola can you share the web page that linked to that image, so i can document it and put it back on the server..

For the other issue run:

```bash
sudo /opt/scripts/tools/version.sh
```
So we can see how your Blue is configured..

Regards,

michaelmegliola, 2018-08-21:
1. The Beagle says:
root@beaglebone:/var/lib/cloud9# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLTBLA21712EL001444]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot
2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot
2017.03-rc2-00002-g11d4fd]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade
<pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video
plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep
admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4
rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.004443] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size
568
dmesg | grep gpio-of-helper

2. The missing debian image appears to be back online, I accessed it from
here (failed earlier today; works now):
[image: image.png]

michaelmegliola, 2018-08-21:
[image]

RobertCNelson, 2018-08-21:
Thanks, @michaelmegliola i added that directory to the whitelist, so those files won't get deleted when i need server space..

For you issue, the old u-boot in the eMMC is blocking things from working correctly:

```bash
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```
Regards,

michaelmegliola, 2018-08-21:
Thanks -- that cleared the bus error.

We are designed engineering projects for high school students using the
BeagleBone Blue... as of now, what version of debian and librobotcontrol
are best to set up a new Beagle?

RobertCNelson, 2018-08-21:
@michaelmegliola right now pretty much any stretch-iot release.

As long as it's booting via: u-boot v2018.03 and v4.14.x-ti based kernel

https://github.com/StrawsonDesign/librobotcontrol/commits/master

Otherwise i'll let @StrawsonDesign give you the set in stone version number..

Regards,

StrawsonDesign, 2018-08-28:
I improved the installation page for clarity and hosted the 8-23 image on my own server so @RobertCNelson doesn't have to worry about keeping images around.

http://strawsondesign.com/docs/librobotcontrol/flashing.html

closed 2018-08-28 by StrawsonDesign

michaelmegliola, 2018-08-28:
Thanks, that is helpful.

StrawsonDesign, 2018-08-28:
Please suggest changes if you think it needs clarification or improvement,
thanks!