beagleboard/librobotcontrol #130: librobotcontrol with RT kernel

rxdu, 2018-09-10:
**Describe the bug**
I'm not sure about the kernel requirements for librobotcontrol. But it seems the library is not fully compatible with the two RT kernels I've tried: bone-rt lts-4.9, lts-4.14. With a freshly installed OS, "rc_test_drivers" can pass all tests. But once the kernel is replaced, some test items will fail. With 4.9rt, the test says pru-rproc driver and LED driver are not loaded.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 2018-8-23 image given at http://www.strawsondesign.com/docs/librobotcontrol/flashing.html
2. Upgrade librobotcontrol with apt-get
3. Update /opt/scripts/tools
4. sudo /opt/scripts/tools/update_kernel.sh --bone-rt-channel --lts-4_9
5. run rc_test_drivers
6. See error

**Expected behavior**
```
Kernel: 4.9.119-bone-rt-r11
BeagleBoard.org Debian Image 2018-08-23
Debian: 9.5

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
ERROR:  pru-rproc driver not loaded
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
ERROR:  LED driver not loaded
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.2
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform information**
Beaglebone Blue, kernel information given above

RobertCNelson, 2018-09-10:
@rxdu the "bone" kernels are not supported by ti's remoteproc pru driver that this library utilizes..

There are 4 u-boot pru options, 3 depend on a specific kernel family..

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_PRU_Options

Regards,

rxdu, 2018-09-10:
@RobertCNelson Thank you for the quick reply. I thought kernels from the bone channel have better customizations and compatibilities with the hardware. I will try RT kernels from the TI channel then.

rxdu, 2018-09-10:
Tested 4.14rt kernel from the TI channel ("4.14.67-ti-rt-r74"). All tests can pass now.

closed 2018-09-10 by rxdu

RobertCNelson, 2018-09-10:
TI is still in the long progress of developing pruss_remoteproc driver, sadly it's been ongoing since v3.14.x and still not in mainline.. Thus not in the bone kernel.  Something "might" finally go into v4.20/v5.0..

Regards,

rxdu, 2018-09-11:
@RobertCNelson May I ask what main differences are between kernels from the two channels? And how are they related to the kernel source files at https://github.com/beagleboard/linux ? Is there a wiki page about this? I didn't find one.

RobertCNelson, 2018-09-11:
@rxdu there's actually 14 channels.. ;)  Most have multiple kernel's..

https://rcn-ee.net/repos/latest/stretch-armhf/

only ti/ti-rt/ti-xenomai have tags on https://github.com/beagleboard/linux

The rest have tags listed here: https://github.com/RobertCNelson/linux-stable-rcn-ee

Regards,

rajmehta28599, 2021-06-04:
> **Describe the bug**
> I'm not sure about the kernel requirements for librobotcontrol. But it seems the library is not fully compatible with the two RT kernels I've tried: bone-rt lts-4.9, lts-4.14. [...]

I have same issue with BeagleBone Black debian image.

pru issue solve with addline in uEnv.txt to 

blacklist pruss
blacklist pruss_intc
blacklist pru-rproc

then reboot
but

ERROR:  LED driver not loaded

is there so i solve this.

jadonk, 2021-06-08:
@rajmehta28599 Thanks for the bug report.

@RobertCNelson Are we still supporting 4.9 or should we just test/fix this on a newer version?

imrj, 2021-07-01:
i have this same issue with both bblue and robotics cape on blackwireless......and using 4.19 kernel