beagleboard/librobotcontrol #131: UART not working on BB Blue

JesperChristensen89, 2018-09-18:
It seems that the UART cannot send/receive anything.
I am able to open each of the available serial ports but no data is sent nor received. 

Are the UARTs simply not available, or have anybody else had similar issues?

RobertCNelson, 2018-09-18:
@JesperChristensen89 lets see your current software stack for debuggin..

Pleas run:

```bash
sudo /opt/scripts/tools/version.sh 
```

Regards

StrawsonDesign, 2018-09-19:
Also please provide the conditions of your test, which port are you using? Also are you testing with a loopback jumper or o-scope? UART is working fine for me right now.

JesperChristensen89, 2018-09-19:
Yes, sorry - I will provide the information below.

I have tested with two seperate BeagleBone Blue Rev A2. Both are freshly flashed with the newest Stretch IoT image, and the librobotcontrol is the newest version from here.

Software stack:
```
git:/opt/scripts/:[73593ebe3b7d3cc381eeb502d45ccb33a6ec5e78]
eeprom:[A335BNLTBLA21712EL001598]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.67-ti-r73]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0~stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.2-git20180829.0-0rcnee0~stretch+20180830]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4,BB-UART5 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.085780] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```

I have tested all of the UARTs both by O-scope and with a loopback jumper. I have both been using the rc_uart_test program, and I have manually opened serial ports tested that.

ttyS0 works (as a serial login console), ttyS1, ttyS2 and ttyS5 are completely silent (no data being sent nor received - verified by o-scope).

JesperChristensen89, 2018-09-19:
Additional observation this morning:

UT5 (UART 5, ttyS5) works just fine. However, still nothing from UT1 or UART 2 from GPS.

StrawsonDesign, 2018-09-19:
Phew, good to know it's just 1&2, that means the pinmux just needs to be set up for those pins. Please try the following and let me know if those ports wake up:

for UART2:
echo uart > /sys/devices/platform/ocp/ocp\:P9_22_pinmux/state
echo uart > /sys/devices/platform/ocp/ocp\:P9_21_pinmux/state

for UART1:
echo uart > /sys/devices/platform/ocp/ocp\:P9_24_pinmux/state
echo uart > /sys/devices/platform/ocp/ocp\:P9_26_pinmux/state

This can also be done with the pinmux module in this library. Since the ports are labeled uart it is probably best that these are in uart mode by default. I've updated the device tree to reflect that with this commit: abf7369df272267db6d9120b78b381bde8db1651

Please clone my repo as it stands now (with this commit included) and install the new device tree as follows:

cd librobotcontrol/device_tree
sudo bash install_dt_manually.sh
select 2 and press enter
sudo reboot

then give it a try, they should work without having to change the pinmux. Let me know your findings, if this works we can update the device tree in the beaglebone images.

closed 2018-09-19 by JesperChristensen89