beagleboard/librobotcontrol #132: Servo module doesn't work as ordinary user

logos-electromechanical, 2018-09-19:
**Describe the bug**
Servo module refuses to initialize when not root

**To Reproduce**
Call rc_servo_init() from a program running as an ordinary user

**Expected behavior**
Servo subsystem should initialize. 

**Screenshots**
Console error message is: 
ERROR: in rc_pru_shared_mem_ptr could not open /dev/mem: Permission denied
Need to be root to access PRU shared memory
ERROR in rc_servo_init, failed to map shared memory pointer

**Platform information**
Debian version is stretch, with apt-get update/upgrade/dist-upgrade run today

uname -a returns:
Linux beaglebone 4.9.45-ti-r57 #1 SMP PREEMPT Fri Aug 25 22:58:38 UTC 2017 armv7l GNU/Linux

rc_test_drivers returns:
Kernel: 4.9.45-ti-r57
BeagleBoard.org Debian Image 2017-08-31
Debian: 9.5

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
ERROR:  ADC iio driver not loaded

logos-electromechanical, 2018-09-19:
Just reflashed with the latest image from scratch, confirmed that this is still an issue with the latest image.

mvduin, 2018-12-28:
Easy solution: use uio-pruss instead of remoteproc-pru. They're both work equally well on rcn's 4.9-ti and 4.14-ti kernels (while 4.9/4.14-bone only support uio-pruss), and using uio lets you avoid having to mess with /dev/mem (and also avoids the bus error if you try to access pru memory while pruss is disabled).

Remoteproc is probably nice if you're writing a kernel driver that interfaces with pru firmware, but uio is superior (imho) when you want to interact directly with pruss from userspace.

Note that if the uio-pruss driver were ever to go away (though I don't see any reason for it since it's trivial to forward-port), you can just as easily use the generic uio_pdrv_genirq driver which has been part of mainline linux since ancient times. The only functionality you'd lose is the shared DDR memory region allocated by uio-pruss, and minor changes would be needed to interrupt handling.

It's unfortunate that libprussdrv is a bit ew (doing dumb shit like hardcoding device paths) and doesn't support loading ELF executables. I still want to write a nicer C/C++ library for uio-pruss based on my experience writing [py-uio](https://github.com/mvduin/py-uio) (which _can_ load both pasm binaries and ELF executables), but haven't gotten around to it yet.