beagleboard/librobotcontrol #133: Heading (compass) calculation error or non-existing in Y-forward-X-up configuration

JesperChristensen89, 2018-09-20:
In our vehicle we need to place the a BeagleBoard Blue in a configuration where the forward is in the Y-direction and up is in the X-direction. 

The MPU-library does not seem to support this configuration and it therefore fails to calculate the heading and compensate correctly for both roll, pitch and yaw. 

I am using the newest image for the BeagleBone Blue along with the newest version (from git) of librobotcontrol. 

Would there be any chance that you could update the libraries to support this, or provide us with instructions how to and then we could submit a PR? 

Additionally we have observed at many occasions during the last year or so that the heading starts drifting (continuing going in circles) all of a sudden, and then needs a hard reboot to reset the IMU. 
Is it possible to avoid or circumvent this when it happens?

StrawsonDesign, 2018-09-27:
I think the orientation you are looking for is the third option, X-UP. You are however right that the magnetometer orientation change was wrong, I fixed that just now in a03bf544d26f74836d8bac78e0d7aa0d3cfde807

I'd love to know more info on the drifting issue, random bugs are hard to track down and this isn't something I've been able to replicate myself.

JesperChristensen89, 2018-09-27:
@StrawsonDesign that was also how I fixed it. And then to use it as compass with Y-forward you futhermore have to multiply with -1 and subtract -90 deg from the result, such that heading becomes `Wrap360(-1*data.compass_heading*Rad2Deg-90)`.

For roll and pitch you then have to grab `data.fused_TaitBryan[TB_ROLL_Y]*Rad2Deg` and `data.fused_TaitBryan[TB_PITCH_X]*Rad2Deg`, respectfully. 

For heading, roll and pitch rate you have to grab `-1*data.gyro[TB_YAW_Z]`, `data.gyro[TB_ROLL_Y]` and `data.gyro[TB_PITCH_X]`.

Just clarifying for anyone else who might end up in this configuration (orientation of BB board). 

In regards to the heading drift, this is something I have seen previously in Sparkfun 9-axis IMUs, which I guess is pretty much the same as the one in the robotics cape. From debugging then, it seemed that if the software utilising the IMU would be restarted several times and thereby re-initiating the IMU several times, the IMU would go crazy. Only fix I could come up with back then was to hard reboot the IMU. But this was only applied after I would have recognized the heading drift.

JesperChristensen89, 2018-09-28:
If they ever found a fix to detect this prior to catastrophic failure and then do a reboot by software, please let me know and share their experiences. 

Thanks,
Jesper