beagleboard/librobotcontrol #140: Can't run robotcontrol service after flash from latest image

anregen, 2018-11-08:
On BBBlue, after flashing using http://strawsondesign.com/docs/librobotcontrol/installation.html and 2018_10_07 image
robotcontrol service does not start and journalctl -xe shows error
```
-- Unit robotcontrol.service has begun starting up.
Nov 08 00:45:34 beaglebone rc_startup_routine[8564]: timeout reached while waiting for pwm driver
Nov 08 00:45:34 beaglebone rc_startup_routine[8564]: TIMEOUT REACHED
Nov 08 00:45:34 beaglebone systemd[1]: robotcontrol.service: Control process exited, code=exited status=255
Nov 08 00:45:34 beaglebone systemd[1]: Failed to start robotcontrol.
-- Subject: Unit robotcontrol.service has failed
```
My devices are
```
debian@beaglebone:/var/lib/cloud9$ ls /sys/class/pwm/
pwmchip0  pwmchip1  pwmchip3  pwmchip4  pwmchip6  pwmchip7
```

but rc_startup_routine.c expects:
```
	if(access("/sys/class/pwm/pwmchip0/export",F_OK)!=0){
		// export directory not there yet, wait a bit more
		return -1;
	}
	if(access("/sys/class/pwm/pwmchip2/export",F_OK)!=0){
		// export directory not there yet, wait a bit more
		return -1;
	}
	if(access("/sys/class/pwm/pwmchip4/export",F_OK)!=0){
		// export directory not there yet, wait a bit more
		return -1;
	}
```
Although robotcontrol service does not run, `rc_test_drivers` passes:
```
debian@beaglebone:/var/lib/cloud9$ rc_test_drivers

Kernel: 4.14.71-ti-r81
BeagleBoard.org Debian Image 2018-10-07
Debian: 9.5

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.3
```

**To Reproduce**
Steps to reproduce the behavior:
1. flash BBBlue with 2018_10_07 image
2. reboot
3. error has occurred
     a) jounalctl -xe or systemctl status robotcontrol to see details of failed operation:
"timeout reached while waiting for pwm driver"
     b) rc_test_drivers still passes all (including PWMs)

My very bad workaround (to see if I could get my CAN code running) was to modify the rc_startup_routine to change pwmchip2 --> pwmchip1.  This "runs", and doesn't immediately throw an error, but I have not tested this to see if PWMs actually behave correctly.  I suspect that they don't, but I'm willing to run some tests on PWMs as needed.  I would need some guidance about how to test.

```
debian@beaglebone:/var/lib/cloud9$ rc_version
1.0.3
debian@beaglebone:/var/lib/cloud9$ uname -a
Linux beaglebone 4.14.71-ti-r80 #1 SMP PREEMPT Fri Oct 5 23:50:11 UTC 2018 armv7l GNU/Linux
```
persists after running /opt/scripts/tools/update_kernel.sh and reboot
```
debian@beaglebone:/var/lib/cloud9$ uname -a
Linux beaglebone 4.14.71-ti-r81 #1 SMP PREEMPT Sat Oct 20 04:20:25 UTC 2018 armv7l GNU/Linux
```

StrawsonDesign, 2018-11-09:
bah, device enumeration bites again. I really wish this popped up before Oct 7th. Anyway, fixed and released as v1.0.4. I released a deb package for the new version, please test and report. https://github.com/StrawsonDesign/librobotcontrol/releases/tag/V1.0.4

closed 2018-11-09 by StrawsonDesign

anregen, 2018-11-09:
After installing this, the service now runs and rc_blink works.
If you give me a bit of a description about what changed and what PWM behavior I'd expect this to effect, I can make sure the PWMs are functional.  e.g. how are the six /sys/class/pwm/pwmchip* files mapped to HW?