beagleboard/librobotcontrol #145: Add support for setting eQEP sampling period

74ls04, 2019-01-07:
**Is your feature request related to a problem? Please describe.**
There is currently no support for changing the sampling period of the eQEP channels. This is supported by the eQEP driver. Setting the period is critical when trying to implement a finely tuned PID controller for a motor. 

**Describe the solution you'd like**
I have implemented and tested this feature in my fork of the repository in [this commit](https://github.com/74ls04/librobotcontrol/commit/21e65f174a31a6de1044c3e3848c27c3ed1a3a00)

Ideally this would have been nice as an argurment in the initializer function but adding one now would break backwards compatibility (without support for default arguments and avoiding messy macros). I opted to create a set and get function. I used the "get" and "set" words in the functions instead of the "read" and "write" that are used for the position -- I think I had a reason for that...

**Describe alternatives you've considered**
Before changing the code I was manually setting the period as such in a bash script. 
`echo "500000" >> /sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/period
`

StrawsonDesign, 2019-01-07:
Nice! I'm sure others will benefit from this. I read through your commit and I really appreciate you keeping the style and doxygen headers consistent. 

Before I merge (and update the docs page) do you mind elaborating in the function descriptions how the system behavior changes? I've never actually played with the period so I can't be sure myself, e.g. do the position reads block until the end of a period? What happens if a period passes without a read, are the counts lost or added to the next read? An example program (or addition to the rc_test_encoders_eqep) demonstrating how to use it properly would really polish this up and make for a nice new library version bump & release.

Let me know what you think. Thanks again, I really like this addition!

74ls04, 2019-01-08:
I'm glad you like it!

The sample period in this context is better understood in relation to the two operating modes, Absolute and Relative. When you poll for the current position in Absolute mode, it will report the new position every **period** nanoseconds. If you don't poll it then the value will keep incrementing or decrementing like normal so this period is just telling it how often to move the latest count into the position count register, which is what we poll.

In Relative mode the period determines how often the position is latched. At each **period** interval, the position is latched, the value is moved to the position count latch register and then the internal counter reset. 

The application of this is that if you're doing RPM/velocity calculations and you want a high "sample rate" (it's really not a true sample rate) then you can adjust that sampling period to correlate with how fast you're polling for the position in order to get more granular position updates. 

Here is a stripped down version of the non-blocking RPM calculation I'm performing in my application.
[image]

Let me convert some of my code into a tachometer sample program that outputs RPMs for the 3 eQEP channels. The inputs will be the encoder resolution (ticks per revolution) and sample period.

I'm using a BeagleBone Black as the sole controller for a robot I'm working on so this library has been a godsend! I'm using pretty much every aspect of it including an IMU and filters.

StrawsonDesign, 2019-01-08:
Excellent, I look forward to playing with your tachometer program. Also I'm very happy to see others making use of the library, it makes all the effort worthwhile!

StrawsonDesign, 2019-03-22:
did you manage to write a tachometer example? I'd love to integrate this into the next version.