beagleboard/librobotcontrol #147: Bring back the manual about decrete time SISO filters

d01010101, 2019-02-05:
Searching for this does not give much but specialized literatur and the manual from StrawsonDesign. Yet you use these in MIP sources. Would it be possible to bring the manual back?

StrawsonDesign, 2019-02-05:
I don't follow. The only manual I wrote on the topic is the primary documentation here

http://strawsondesign.com/docs/librobotcontrol/group___s_i_s_o___filter.html

d01010101, 2019-02-05:
Yes, it is preliminary, and I hoped for some continuation. Google shows this:

www.strawsondesign.com/content/manual-discrete-siso
Discrete-Time SISO Filters In this modern age of low cost microcontrollers its is now the norm to control continuous time dynamic systems with ...

but in reality this page does not exist. Of course you can find generic manuals like https://www.ethz.ch/content/dam/ethz/special-interest/mavt/dynamic-systems-n-control/idsc-dam/Lectures/Digital-Control-Systems/Slides_DigReg_2013.pdf but I thought that you have something more specific to the library, usage of buffers etc.

If it never existed, I will just try to use the API docs.

StrawsonDesign, 2019-02-05:
Ah yes, I remeber that page now, thats was 3-4 years ago and was very brief. The new doxygen docs are many times more detailed.

I highly suggest Thomas Bewley's book, Numerical Renaissance, for learning about SISO filters. It's available as a free PDF at http://numerical-renaissance.com/

closed 2019-02-05 by StrawsonDesign

StrawsonDesign, 2019-02-05:
Correct, although if you want more history you can use rc_ringbuf_alloc to allocate more space in the rc_filter_t struct input/output buffers without affecting filter behaviour.

StrawsonDesign, 2019-02-05:
What would you want to tinker with? Perhaps there is an easy feature I could add.

d01010101, 2019-02-06:
I would not want to tinker, it is the advantage of these filters that they work out of the box. When I initially read something about handling buffers, I was afraid that I need to tinker more than normally with filters like that and I asked about docs. It is a good set of classic filters, which has all the features I want.