beagleboard/librobotcontrol #149: set thread attributes for rc_mpu_set_dmp_callback()

d01010101, 2019-02-20:
Possibly I am wrong, but I have an impression, that creating a real-time thread causes calls via  rc_mpu_set_dmp_callback() to starve. What is the thread which makes these calls? Can the thread be made real time?

StrawsonDesign, 2019-03-22:
the dmp callbacks are executed in the thread started here: https://github.com/StrawsonDesign/librobotcontrol/blob/master/library/src/mpu/mpu.c#L962 with the schedule policy and priority specified in the rc_mpu_config_t struct letting the user set any policy (real time FIFO or other) that they like.

Yes, I believe if the user sets real time FIFO scheduling for this thread and another FIFO thread (even in another process) to a higher priority that locks the CPU this thread could be starved. That's up to the user to decide which scheduler is right for their program. For a discrete feedback controller running off the IMU, executing the feedback in this callback with a high priority is reasonable and has been working well for me for years.

I hope this helps, let me know if I should explain further or if I missed the plot completely.

d01010101, 2019-03-22:
Yes it helps completely, thanks.

closed 2019-03-22 by d01010101