beagleboard/librobotcontrol #150: robotcontrol_modules.conf is bogus

mvduin, 2019-02-26:
When testing a 4.19-ti kernel I noticed on my system that systemd-modules-load.service had failed at boot.

A quick inspection of the culprit, `/etc/modules-load.d/robotcontrol_modules.conf`, revealed that this file is entirely pointless. All of the modules listed there are for drivers that are found based on of_device_id and they will be loaded automatically if and when they are needed (if compiled as module of course).

Indeed, after removing this file, all of the modules listed (apart from the two builtins) are still loaded during boot on 4.14-ti with cape-universal and remoteproc-pru enabled (to ensure there are devices that require these drivers).

StrawsonDesign, 2019-02-26:
mmm kind of. You are right that all of the required modules will load eventually, the problem is when. I used to have the robotcontrol systemd service wait for gpio/pwm/eqep/pru drivers to be loaded before starting, but this would take 20+ seconds. With the modules.conf file systemd makes sure those modules are loaded before the robotcontrol service starts which shaved almost 10 seconds off the time from power-on to the robotcontrol service starting. 

I'm more curious what's failing in 4.19 and fixing that. Can you provide more info on the exact problem you see in 4.19?

RobertCNelson, 2019-02-26:
@StrawsonDesign btw, many of these default modules have been moved to built-in to help speed up boot-up time for common setup's..

```
CONFIG_PWM_TIECAP=y
CONFIG_PWM_TIEHRPWM=y
CONFIG_PWM_TIPWMSS=y
CONFIG_TIEQEP=y
```

Just spidev/remoteproc remain as modules:

```
CONFIG_SPI_SPIDEV=m
CONFIG_OMAP_REMOTEPROC=m
```

I think we had issues back in early v4.4.x-ti days with some modules not correctly loading purely on dt nodes.  But I agree with @mvduin with v4.14.x/v4.19.x most of those issues are now fixed.  For our current images, stretch (v4.14.x), buster (v4.19.x+) and bionic (v4.19.x+), this would be safe for the robotcontrol package.

Regards,

mvduin, 2019-02-27:
> I used to have the robotcontrol systemd service wait for gpio/pwm/eqep/pru drivers to be loaded before starting, but this would take 20+ seconds

Odd, why do those drivers get loaded so late? Does this issue also exist when not using initramfs?

> shaved almost 10 seconds off the time from power-on to the robotcontrol service starting

Well, for that there's more low-hanging fruit... on a mostly-unmodified stretch-iot 2018-10-07 image (booting from sd card) I did a few tests of startup time, since I was surprised that updating the bb-customizations package (with new udev rules) didn't help much:

- 1m25s initially
- 48s after rebuilding initramfs (to include the updated udev rules)
- 36s after removing initramfs entirely (there's no reason to use one)

(with cape-universal disabled those three numbers are 38s, 36s, 28s)

mvduin, 2019-02-27:
> I'm more curious what's failing in 4.19 and fixing that. Can you provide more info on the exact problem you see in 4.19?

There's no remoteproc-pru yet, only uio-pruss.

RobertCNelson, 2019-02-27:
1m25s initially - silly ssh key's generation messing up the boot log..

remoteproc_pruss, just landed last night, 4.19.25-ti-r11, i'll be pushing out a new overlay shortly..

Regards,

mvduin, 2019-02-27:
No I mean initially when I started these tests to figure out why updating the bb-customizations package didn't help (the reason of which was that initramfs didn't get rebuilt hence was still using the udev rules). I didn't mean on first boot.

RobertCNelson, 2019-02-27:
So the Blue this is toss up:

Default:
```
debian@beaglebone:~$ systemd-analyze 
Startup finished in 5.677s (kernel) + 32.380s (userspace) = 38.058s
debian@beaglebone:~$ systemd-analyze blame | grep robotcontrol
          1.214s robotcontrol.service
```

Removed: /etc/modules-load.d/robotcontrol_modules.conf
```
debian@beaglebone:~$ systemd-analyze 
Startup finished in 5.738s (kernel) + 33.257s (userspace) = 38.996s
debian@beaglebone:~$ systemd-analyze blame | grep robotcontrol
         20.626s robotcontrol.service
```

StrawsonDesign, 2019-02-27:
Thanks for posting those results, @RobertCNelson :) Is this with 4.19.25-ti-r11?

Total startup time seems to remain ~38 seconds, but with modules.conf the robotcontrol service started 19.4 seconds sooner. 19 extra seconds is a long time to wait for a robot/quadcopter to start after plugging it in.

RobertCNelson, 2019-02-27:
My quick test is with 4.14.94-ti-r94.. I'm going to test 4.19.x-ti shortly as i just got the TI remotepruss working last night..

But with v4.14.x-ti, we can see it doesn't matter right now who loads the module, the boot time is the same..

Okay, now let's test v4.19.x..

Regards,

mvduin, 2019-02-27:
what happens without initramfs? (just remove/rename the `/boot/initrd-$(uname -r)` file or whatever it's called)

StrawsonDesign, 2019-02-27:
Total boot time may be the same, but robots can start moving/balancing/flying before other services like apaceh2 and cloud9 are up. On our last stable image the MiP reference design would start balancing 12 to 15 seconds (if I remember correctly) after connecting power which is acceptable. pruss-remoteproc is still a dependency for major librobotcontrol functions so I'd like to keep it in modules.conf considering it seems to have just arrived in 4.19

RobertCNelson, 2019-02-27:
Switching back to the Blue, and i'll grab the svg plots so we can see when librobotcontrol finished..

Regards,

StrawsonDesign, 2019-02-27:
thanks @RobertCNelson sorry I'm at work right now without a beaglebone to test.

RobertCNelson, 2019-02-27:
[image]

StrawsonDesign, 2019-02-27:
do you mind having a look at /var/log/robotcontrol/startup_log.txt too? (sorry) if any required modules fail to load before the robotcontrol service starts they will be logged there with timestamp.

RobertCNelson, 2019-02-27:
So the graphs are built into systemd:

systemd-analyze plot > /var/www/html/no_intrd.svg

Then i just used https://svgtopng.com to convert them for github.

Regards,

StrawsonDesign, 2019-02-27:
Neat, I'll definitely use that feature in the future! As usual, thanks for the help @RobertCNelson ! Closing this for now.

closed 2019-02-27 by StrawsonDesign

mvduin, 2019-02-27:
At least consider adding a comment to the file explaining why it exists

StrawsonDesign, 2019-02-27:
no problem: 3d80f6222db98ad0894e4bff37bdb801b8fe5497

RobertCNelson, 2019-02-27:
So if we build it in:
```
debian@beaglebone:~$ systemd-analyze 
Startup finished in 1.508s (kernel) + 32.830s (userspace) = 34.338s
debian@beaglebone:~$ systemd-analyze blame | grep robo
          1.213s robotcontrol.service
debian@beaglebone:~$ cat  /var/log/robotcontrol/startup_log.txt 
start time (s):   7.83
time (s): 8.36 GPIO loaded
time (s): 8.41 PWM loaded
time (s): 8.46 eQEP loaded
time (s): 8.51 PRU loaded
startup routine complete
```

```
diff --git a/patches/defconfig b/patches/defconfig
index 2572c2a..4986ccb 100644
--- a/patches/defconfig
+++ b/patches/defconfig
@@ -2891,7 +2891,7 @@ CONFIG_SPI_TI_QSPI=y
 #
 # SPI Protocol Masters
 #
-CONFIG_SPI_SPIDEV=m
+CONFIG_SPI_SPIDEV=y
 # CONFIG_SPI_LOOPBACK_TEST is not set
 # CONFIG_SPI_TLE62X0 is not set
 # CONFIG_SPI_SLAVE is not set
@@ -5354,20 +5354,20 @@ CONFIG_OMAP_IOMMU=y
 # Remoteproc drivers
 #
 CONFIG_REMOTEPROC=y
-CONFIG_OMAP_REMOTEPROC=m
+CONFIG_OMAP_REMOTEPROC=y
 # CONFIG_OMAP_REMOTEPROC_WATCHDOG is not set
 CONFIG_WKUP_M3_RPROC=y
-CONFIG_PRUSS_REMOTEPROC=m
+CONFIG_PRUSS_REMOTEPROC=y
 
 #
 # Rpmsg drivers
 #
-CONFIG_RPMSG=m
+CONFIG_RPMSG=y
 CONFIG_RPMSG_CHAR=m
 # CONFIG_RPMSG_QCOM_GLINK_RPM is not set
-CONFIG_RPMSG_VIRTIO=m
-CONFIG_RPMSG_RPC=m
-CONFIG_RPMSG_PRU=m
+CONFIG_RPMSG_VIRTIO=y
+CONFIG_RPMSG_RPC=y
+CONFIG_RPMSG_PRU=y
 
 #
 # SOC (System On Chip) specific Drivers
```

it still "rc_balance".. ;)  maybe with v4.19.x-ti we can try it built in prior.. (v4.14.x-ti still had a back door for kernel overlays, with v4.19.x-ti that's all gone..)

Regards,