beagleboard/librobotcontrol #151: [WIP]: v4.19.x-ti remoteproc_pruss changes...

RobertCNelson, 2019-02-26:
So, TI's v4.19.x-ti branch just landed TI's remoteproc_pruss changes a few hours ago:

http://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=8de817772462078a064b70d368eeae3b92a31439

Right now i don't have a kernel to boot/test, but the pru node changed again, here is the diff:

```
diff --git a/4.14.x.pru.txt b/4.19.x.pru.txt
index 6e7df3c..6be9822 100644
--- a/4.14.x.pru.txt
+++ b/4.19.x.pru.txt
@@ -1,15 +1,15 @@
-		pruss_soc_bus: pruss_soc_bus@4a326004 {
+		pruss_soc_bus: pruss-soc-bus@4a326004 {
 			compatible = "ti,am3356-pruss-soc-bus";
 			reg = <0x4a326004 0x4>;
 			ti,hwmods = "pruss";
 			#address-cells = <1>;
 			#size-cells = <1>;
-			ranges = <0x0 0x4a300000 0x80000>;
+			ranges;
 			status = "disabled";
 
-			pruss: pruss@0 {
+			pruss: pruss@4a300000 {
 				compatible = "ti,am3356-pruss";
-				reg = <0x0 0x80000>;
+				reg = <0x4a300000 0x80000>;
 				interrupts = <20 21 22 23 24 25 26 27>;
 				interrupt-names = "host2", "host3", "host4",
 						  "host5", "host6", "host7",
@@ -19,42 +19,41 @@
 				ranges;
 				status = "disabled";
 
-				pruss_mem: memories@0 {
-					reg = <0x0 0x2000>,
-					      <0x2000 0x2000>,
-					      <0x10000 0x3000>;
+				pruss_mem: memories@4a300000 {
+					reg = <0x4a300000 0x2000>,
+					      <0x4a302000 0x2000>,
+					      <0x4a310000 0x3000>;
 					reg-names = "dram0", "dram1",
 						    "shrdram2";
 				};
 
-				pruss_cfg: cfg@26000 {
+				pruss_cfg: cfg@4a326000 {
 					compatible = "syscon";
-					reg = <0x26000 0x2000>;
+					reg = <0x4a326000 0x2000>;
 				};
 
-				pruss_iep: iep@2e000 {
+				pruss_iep: iep@4a32e000 {
 					compatible = "syscon";
-					reg = <0x2e000 0x31c>;
+					reg = <0x4a32e000 0x31c>;
 				};
 
-				pruss_mii_rt: mii_rt@32000 {
+				pruss_mii_rt: mii-rt@4a332000 {
 					compatible = "syscon";
-					reg = <0x32000 0x58>;
+					reg = <0x4a332000 0x58>;
 				};
 
-				pruss_intc: intc@20000 {
+				pruss_intc: interrupt-controller@4a320000 {
 					compatible = "ti,am3356-pruss-intc";
-					reg = <0x20000 0x2000>;
-					reg-names = "intc";
+					reg = <0x4a320000 0x2000>;
 					interrupt-controller;
 					#interrupt-cells = <1>;
 				};
 
-				pru0: pru@34000 {
+				pru0: pru@4a334000 {
 					compatible = "ti,am3356-pru";
-					reg = <0x34000 0x2000>,
-					      <0x22000 0x400>,
-					      <0x22400 0x100>;
+					reg = <0x4a334000 0x2000>,
+					      <0x4a322000 0x400>,
+					      <0x4a322400 0x100>;
 					reg-names = "iram", "control", "debug";
 					firmware-name = "am335x-pru0-fw";
 					interrupt-parent = <&pruss_intc>;
@@ -62,11 +61,11 @@
 					interrupt-names = "vring", "kick";
 				};
 
-				pru1: pru@38000 {
+				pru1: pru@4a338000 {
 					compatible = "ti,am3356-pru";
-					reg = <0x38000 0x2000>,
-					      <0x24000 0x400>,
-					      <0x24400 0x100>;
+					reg = <0x4a338000 0x2000>,
+					      <0x4a324000 0x400>,
+					      <0x4a324400 0x100>;
 					reg-names = "iram", "control", "debug";
 					firmware-name = "am335x-pru1-fw";
 					interrupt-parent = <&pruss_intc>;
@@ -74,9 +73,9 @@
 					interrupt-names = "vring", "kick";
 				};
 
-				pruss_mdio: mdio@32400 {
+				pruss_mdio: mdio@4a332400 {
 					compatible = "ti,davinci_mdio";
-					reg = <0x32400 0x90>;
+					reg = <0x4a332400 0x90>;
 					clocks = <&dpll_core_m4_ck>;
 					clock-names = "fck";
 					bus_freq = <1000000>;
```

Here is the current User Space diff:

```
empty - will fill in a day or two...
```

Regards,

RobertCNelson, 2019-03-04:
So with, 4.19.25-ti-r12 rc_balance just works. ;)

```
Kernel: 4.19.25-ti-r12
BeagleBoard.org Debian Image 2019-02-24
Debian: 9.8

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
PASSED: pwm0
PASSED: pwm1
PASSED: pwm2
PASSED: eqep0
PASSED: eqep1
PASSED: eqep2
PASSED: pru-rproc
PASSED: uart1
PASSED: uart2
PASSED: uart4
PASSED: uart5
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLUE
Robot Control library Version:
1.0.4
debian@beaglebone:~$ rc_balance 
```

StrawsonDesign, 2019-03-08:
wow!!!! great news and very well done with the 4.19 work. I'll make a note that this is a working kernel revision.

closed 2019-03-08 by StrawsonDesign