beagleboard/librobotcontrol #152: MPU initialize failed

kambic, 2019-02-28:
hello.
When I try test MPU, it fails to initialize, also in dmp mode. i get:

root@arm:/home/ubuntu# rc_test_mpu 
ERROR: in rc_i2c_write_byte, system write returned -1, expected 2
ERROR: in rc_i2c_write_byte, system write returned -1, expected 2
ERROR resetting MPU, I2C write to reset bit failed
ERROR: failed to reset_mpu9250
rc_mpu_initialize_failed
root@arm:/home/ubuntu# 

runing ubuntu image:
root@arm:/home/ubuntu# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

root@arm:/home/ubuntu# uname -a
Linux arm 4.14.79-ti-r84 #1 SMP PREEMPT Tue Nov 13 20:35:19 UTC 2018 armv7l armv7l armv7l GNU/Linux
root@arm:/home/ubuntu# 

I use robotics-cape on BB black.

RobertCNelson, 2019-02-28:
@kambic please provide the output of:

sudo /opt/scripts/tools/version.sh

Regards,

kambic, 2019-03-01:
Hello. Below is my output. I had problems with IMU from first day. sometimes it works sometimes it not. Yesterday after few hours of trying i found out if i head robotisc cape with hair-dryer it start working, until the board cools down. 
 
sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[0ca42da8c7e3d81e9cbab71b1e2b0fad00bf79ca]
eeprom:[A335BNLTBWA51650BBWG1785]
model:[TI_AM335x_BeagleBone_Black_Wireless_RoboticsCape]
dogtag:[rcn-ee.net console Ubuntu Image 2018-09-11]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00003-gaa563917cb]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.94-ti-r94]
device-tree-override:[dtb=am335x-boneblack-wireless-roboticscape.dtb]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190103.0-0rcnee0~bionic+20190103]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~bionic+20180517]
pkg:[kmod]:[24-1ubuntu3.1rcnee0~bionic+20181120]
pkg:[librobotcontrol]:[1.0.4-git20190103.0-0rcnee0~bionic+20190104]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M]
dmesg | grep pinctrl-single
[    1.042393] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper

StrawsonDesign, 2019-03-22:
Looks like you got a dud IMU, sorry!

closed 2019-03-22 by StrawsonDesign