beagleboard/librobotcontrol #154: motor command results in "segmentation fault"

dmalawey, 2019-03-20:
**Describe the bug**
Driving motors on beaglebone blue generates a "segmentation fault" after one or more iterations.  After the fault, the beaglebone does not accept new commands.  A sample size of 10 beaglebones have been used and about 3 or 4 of them repeatedly exhibit this problem.   Rebooting the board always fixes the problem for at least one run of the command, then the problem returns.

**To Reproduce**
Steps to reproduce the behavior:
1. run rc_test_motors -d 1 -m 1 or make any command for operating motors.
2. ctrl-c to stop running. (at this time, the fault may be generated)
3. attempt to run another motor command 

**Expected behavior**
rc_test_motors command should result in the same output every time it is run, and shut down cleanly.  ctrl-c and running the command again should run the motors again.

**Screenshots**

```
debian@192.168.8.1's password:
Last login: Fri Mar  1 22:52:19 2019 from 192.168.8.50
➜  ~ rc_test_motors

-d {duty}   define a duty cycle from -1.0 to 1.0
-b          enable motor brake function
-F {freq}   set a custom pwm frequency in HZ, otherwise default 25000 is used
-f          enable free spin function
-s {duty}   sweep motors back and forward at duty cycle
-m {motor}  specify a single motor from 1-4, otherwise all will be driven
            motors will be driven equally.
-h          print this help message

➜  ~ rc_test_motors -d 1 -m 1
sending duty cycle 1.0000
^C
calling rc_motor_cleanup()

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.337703] Internal error: : 1028 [#1] PREEMPT SMP ARM

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.455580] Process rc_test_motors (pid: 1228, stack limit = 0xdafa82                                                          18)

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.462311] Stack: (0xdafa9e28 to 0xdafaa000)

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.466691] 9e20:                   dafa9e7c db49be80 00009c40 db49be                                                          a0 00000000 00000000

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.474909] 9e40: dafa9e74 dafa9e50 c081b770 c081e234 c1504dc8 000000                                                          00 daaa1794 db49be80

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.483126] 9e60: 00000002 daf48700 dafa9eb4 dafa9e78 c081c008 c081b6                                                          18 00000000 00009c40

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.491343] 9e80: 00000000 00000000 00000000 322c3fc2 00000002 c081bf                                                          60 dc4df780 00000000

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.499560] 9ea0: 00000000 dafa9f68 dafa9ecc dafa9eb8 c092f108 c081bf                                                          6c c092f0e0 dc4df780

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.507777] 9ec0: dafa9ee4 dafa9ed0 c03855c4 c092f0ec 00000002 dc4df7                                                          80 dafa9f1c dafa9ee8

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.515995] 9ee0: c0384c10 c0385580 00000000 00000000 db701d88 c0384b                                                          18 db7c0a80 b6e5caec

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.524211] 9f00: dafa9f68 00000000 b6e5caec 00000002 dafa9f34 dafa9f                                                          20 c02f9258 c0384b24

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.532428] 9f20: 00000002 db7c0a80 dafa9f64 dafa9f38 c02f943c c02f92                                                          3c 00000000 c031bba0

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.540645] 9f40: c1504dc8 db7c0a80 00000000 00000000 db7c0a80 b6e5ca                                                          ec dafa9fa4 dafa9f68

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.548862] 9f60: c02f96a4 c02f9394 00000000 00000000 00000002 322c3f                                                          c2 00000005 00000003

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.557079] 9f80: 00000002 00000004 00000004 c01090e4 dafa8000 000000                                                          00 00000000 dafa9fa8

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.565296] 9fa0: c0108f00 c02f9654 00000003 00000002 00000003 b6e5ca                                                          ec 00000002 b6e7d624

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.573512] 9fc0: 00000003 00000002 00000004 00000004 00000000 000000                                                          00 004b1000 00000000

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.581730] 9fe0: 00000000 bea7a4cc b6e3f14d b6e1e3b6 00010030 000000                                                          03 00000000 00000000

Message from syslogd@scuttle at Mar  1 22:42:44 ...
 kernel:[  408.660303] Code: 03e06003 13e0600c 03a07001 13a07004 (e1d340b0)
[1]    1228 segmentation fault  rc_test_motors -d 1 -m 1
```

**Platform information**
Linux version 4.14.71-ti-r80 (root@b2-am57xx-beagle-x15-2gb) (gcc             version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)) #1 SMP PREEMPT Fri Oct 5 23:50            :11 UTC 2018

jadonk, 2019-03-26:
Trying to develop a test case for the issue, but I've run into a different one. This is after seeing a segmentation fault.

```
debian@beaglebone:/var/lib/cloud9$ bash <<EOF
set -x
set -e
while true; do
rc_test_motors -d 0.2 -m 1 &
RC=$!
sleep 1
kill -15 \$RC
sleep 1
done
EOF
+ set -e
+ true
+ RC=
+ sleep 1
+ rc_test_motors -d 0.2 -m 1
ERROR in rc_pwm_init, export failed for subsystem 1 channel 0
tried accessing /sys/class/pwm/pwm-4:0/enable
ERROR in rc_motor_init, failed to initialize pwm subsystem 1
+ kill -15
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
```

jadonk, 2019-03-26:
I'm getting segmentation faults as I try to unload the PWM drivers. See https://gist.github.com/jadonk/8ece4ad83ac67e5215af6a3d686d0ae2

```sh
debian@beaglebone:/var/lib/cloud9$ sudo ./pwmcleaner.sh
+ set -e
++ ls -d /sys/class/pwm/pwm-4:1
+ FILES=/sys/class/pwm/pwm-4:1
+ for file in $FILES
+ echo Found /sys/class/pwm/pwm-4:1
Found /sys/class/pwm/pwm-4:1
++ echo /sys/class/pwm/pwm-4:1
++ cut -d- -f 2
+ PWM=4:1
++ echo 4:1
++ cut -d: -f 1
+ PWMCHIP=pwmchip4
++ cut -d: -f 2
++ echo 4:1
+ CHANNEL=1
+ echo Disabling pwmchip4 channel 1...
Disabling pwmchip4 channel 1...
+ echo 0
./pwmcleaner.sh: line 11: /sys/class/pwm//sys/class/pwm/pwm-4:1/enable: No such file or directory
debian@beaglebone:/var/lib/cloud9$ sudo ./pwmcleaner.sh
+ set -e
++ ls -d pwm-4:1
+ FILES=pwm-4:1
+ for file in $FILES
+ echo Found pwm-4:1
Found pwm-4:1
++ echo pwm-4:1
++ cut -d- -f 2
+ PWM=4:1
++ echo 4:1
++ cut -d: -f 1
+ PWMCHIP=pwmchip4
++ cut -d: -f 2
++ echo 4:1
+ CHANNEL=1
+ echo Disabling pwmchip4 channel 1...
Disabling pwmchip4 channel 1...
+ echo 0

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.327770] Internal error: : 1028 [#2] PREEMPT SMP ARM

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.446786] Process pwmcleaner.sh (pid: 1696, stack limit = 0xdafbe218)

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.453429] Stack: (0xdafbfe28 to 0xdafc0000)

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.457810] fe20:                   dafbfe7c db422eb0 00009c40 db422ed0 00000000 00000000

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.466028] fe40: dafbfe74 dafbfe50 c081b770 c081e234 c1504dc8 00000000 daadaf94 db422eb0

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.474246] fe60: 00000002 daffa800 dafbfeb4 dafbfe78 c081c008 c081b618 00000000 00009c40

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.482463] fe80: 00000000 00000000 00000000 322c3fc2 00000002 c081bf60 db6e6980 00000000

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.490680] fea0: 00000000 dafbff68 dafbfecc dafbfeb8 c092f108 c081bf6c c092f0e0 db6e6980

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.498898] fec0: dafbfee4 dafbfed0 c03855c4 c092f0ec 00000002 db6e6980 dafbff1c dafbfee8

[... 68 lines cut ...]
[ 1731.474246] fe60: 00000002 daffa800 dafbfeb4 dafbfe78 c081c008 c081b618 00000000 00009c40
[ 1731.482463] fe80: 00000000 00000000 00000000 322c3fc2 00000002 c081bf60 db6e6980 00000000
[ 1731.490680] fea0: 00000000 dafbff68 dafbfecc dafbfeb8 c092f108 c081bf6c c092f0e0 db6e6980
Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.564635] ffc0: 00000002 00926a08 b6f69d60 00000004 00926a08 00000002 00000000 00000000

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.572853] ffe0: 00000000 bedd5fec b6ecc1bb b6f08af6 00000030 00000001 00000000 00000000

Message from syslogd@beaglebone at Mar 26 15:42:29 ...
 kernel:[ 1731.651426] Code: 03e06003 13e0600c 03a07001 13a07004 (e1d340b0)
Segmentation fault
debian@beaglebone:/var/lib/cloud9$ dmesg | tail
[ 1731.589657] [<c081b770>] (pwm_apply_state) from [<c081c008>] (enable_store+0xa8/0xd4)
[ 1731.597530] [<c081c008>] (enable_store) from [<c092f108>] (dev_attr_store+0x28/0x34)
[ 1731.605320] [<c092f108>] (dev_attr_store) from [<c03855c4>] (sysfs_kf_write+0x50/0x54)
[ 1731.613279] [<c03855c4>] (sysfs_kf_write) from [<c0384c10>] (kernfs_fop_write+0xf8/0x1d4)
[ 1731.621504] [<c0384c10>] (kernfs_fop_write) from [<c02f9258>] (__vfs_write+0x28/0x48)
[ 1731.629375] [<c02f9258>] (__vfs_write) from [<c02f943c>] (vfs_write+0xb4/0x1c0)
[ 1731.636722] [<c02f943c>] (vfs_write) from [<c02f96a4>] (SyS_write+0x5c/0xbc)
[ 1731.643815] [<c02f96a4>] (SyS_write) from [<c0108f00>] (ret_fast_syscall+0x0/0x54)
[ 1731.651426] Code: 03e06003 13e0600c 03a07001 13a07004 (e1d340b0)
[ 1731.657554] ---[ end trace 9791c21ea393bf4b ]---
debian@beaglebone:/var/lib/cloud9$ dmesg | less
```

jadonk, 2019-03-26:
Testing updated kernel:

```sh
debian@beaglebone:/var/lib/cloud9$ cd 8ece4ad83ac67e5215af6a3d686d0ae2/
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$ ls
pwmcleaner.sh  README.md  thrash_motors.sh
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$ sudo ./thrash_motors.sh
[sudo] password for debian:
+ set -e
+ true
+ RC=1143
+ sleep 1
+ rc_test_motors -d 0.2 -m 1
sending duty cycle 0.2000
+ kill -15 1143
+ sleep 1
./thrash_motors.sh: line 10:  1143 Terminated              rc_test_motors -d 0.2 -m 1
+ true
+ RC=1185
+ sleep 1
+ rc_test_motors -d 0.2 -m 1
sending duty cycle 0.2000
+ kill -15 1185
+ sleep 1
./thrash_motors.sh: line 10:  1185 Terminated              rc_test_motors -d 0.2 -m 1
+ true
+ RC=1227
+ sleep 1
+ rc_test_motors -d 0.2 -m 1

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.514745] Internal error: : 1028 [#1] PREEMPT SMP ARM

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.636042] Process rc_test_motors (pid: 1227, stack limit = 0xdaf9e218)

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.642771] Stack: (0xdaf9fe18 to 0xdafa0000)

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.647149] fe00:                                                       00000001 00000000

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.655367] fe20: 00000000 daf9fe7c db492400 c075dbd8 db492420 00000000 00000000 daa14310

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.663585] fe40: daf9fe74 daf9fe50 c075b3b8 c075dbe4 c1504dc8 00000000 db7dc594 db492400

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.671802] fe60: 00000002 dab6d600 daf9feb4 daf9fe78 c075bb08 c075b2a4 00000000 00009c40

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.680020] fe80: 00000000 00000000 00000001 322c3fe2 db63f210 c075ba8c daa14300 00000000

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.688237] fea0: 00000000 daf9ff68 daf9fecc daf9feb8 c086f318 c075ba98 c086f2f0 daa14300

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.696454] fec0: daf9fee4 daf9fed0 c03842a0 c086f2fc 00000002 daa14300 daf9ff1c daf9fee8

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.704671] fee0: c03838ec c038425c 00000000 00000000 db630d88 c03837f4 dc589480 b6ef1aec

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.712888] ff00: daf9ff68 00000000 b6ef1aec 00000002 daf9ff34 daf9ff20 c02f9db4 c0383800

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.721105] ff20: 00000002 dc589480 daf9ff64 daf9ff38 c02f9fb8 c02f9d98 00000000 c031c82c

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.729323] ff40: c1504dc8 dc589480 00000000 00000000 dc589480 b6ef1aec daf9ffa4 daf9ff68

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.737540] ff60: c02fa230 c02f9f10 00000000 00000000 00000002 322c3fe2 00000005 00000001

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.745757] ff80: 0000000a 00000000 00000004 c0108f44 daf9e000 00000004 00000000 daf9ffa8

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.753973] ffa0: c0108d40 c02fa1e0 00000001 0000000a 00000003 b6ef1aec 00000002 b6f12620

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.762190] ffc0: 00000001 0000000a 00000000 00000004 bed7b518 00000005 00000006 00000007

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.770406] ffe0: 00000000 bed7b45c b6ed3c8d b6eb33b6 00070030 00000003 00000000 00000000

Message from syslogd@beaglebone at Mar 26 16:20:45 ...
 kernel:[  182.849601] Code: e5980000 e6ffa07e eb0494a7 e5985028 (e1d570b0)
./thrash_motors.sh: line 10:  1227 Segmentation fault      rc_test_motors -d 0.2 -m 1
+ kill -15 1227
./thrash_motors.sh: line 8: kill: (1227) - No such process
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$ dmesg | pastebin
bash: pastebin: command not found
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$ dmesg | pastebinit
http://paste.debian.net/1074786/
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$
```

jadonk, 2019-03-26:
Verified version 1.0.4 also fails. I'll try some patches to the export/unexport code to see if it can be worked-around.

```sh
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$ sudo ./thrash_motors.sh
[sudo] password for debian:
+ set -e
+ true
+ RC=1132
+ sleep 1
+ rc_test_motors -d 0.2 -m 1
sending duty cycle 0.2000
+ kill -15 1132
+ sleep 1
./thrash_motors.sh: line 10:  1132 Terminated              rc_test_motors -d 0.2 -m 1
+ true
+ RC=1174
+ sleep 1
+ rc_test_motors -d 0.2 -m 1
sending duty cycle 0.2000
+ kill -15 1174
+ sleep 1
./thrash_motors.sh: line 10:  1174 Terminated              rc_test_motors -d 0.2 -m 1
+ true
+ RC=1216
+ sleep 1
+ rc_test_motors -d 0.2 -m 1

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.347134] Internal error: : 1028 [#1] PREEMPT SMP ARM

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.468431] Process rc_test_motors (pid: 1216, stack limit = 0xdafa0218)

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.475161] Stack: (0xdafa1e18 to 0xdafa2000)

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.479539] 1e00:                                                       00000001 00000000

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.487757] 1e20: 00000000 dafa1e7c db491900 c075dbd8 db491920 00000000 00000000 db17fb90

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.495974] 1e40: dafa1e74 dafa1e50 c075b3b8 c075dbe4 c1504dc8 00000000 dab10594 db491900

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.504190] 1e60: 00000002 daf71740 dafa1eb4 dafa1e78 c075bb08 c075b2a4 00000000 00009c40

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.512408] 1e80: 00000000 00000000 00000001 322c3fe2 db63e010 c075ba8c db17fb80 00000000

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.520625] 1ea0: 00000000 dafa1f68 dafa1ecc dafa1eb8 c086f318 c075ba98 c086f2f0 db17fb80

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.528842] 1ec0: dafa1ee4 dafa1ed0 c03842a0 c086f2fc 00000002 db17fb80 dafa1f1c dafa1ee8

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.537058] 1ee0: c03838ec c038425c 00000000 00000000 db631d88 c03837f4 dc3b8180 b6f10aec

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.545276] 1f00: dafa1f68 00000000 b6f10aec 00000002 dafa1f34 dafa1f20 c02f9db4 c0383800

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.553494] 1f20: 00000002 dc3b8180 dafa1f64 dafa1f38 c02f9fb8 c02f9d98 00000000 c031c82c

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
 kernel:[  234.561712] 1f40: c1504dc8 dc3b8180 00000000 00000000 dc3b8180 b6f10aec dafa1fa4 dafa1f68

Message from syslogd@beaglebone at Mar 26 16:33:05 ...
[... 425 lines cut ...]
[  234.512408] 1e80: 00000000 00000000 00000001 322c3fe2 db63e010 c075ba8c db17fb80 00000000
[  234.520625] 1ea0: 00000000 dafa1f68 dafa1ecc dafa1eb8 c086f318 c075ba98 c086f2f0 db17fb80
[  234.528842] 1ec0: dafa1ee4 dafa1ed0 c03842a0 c086f2fc 00000002 db17fb80 dafa1f1c dafa1ee8
[  234.537058] 1ee0: c03838ec c038425c 00000000 00000000 db631d88 c03837f4 dc3b8180 b6f10aec
[  234.545276] 1f00: dafa1f68 00000000 b6f10aec 00000002 dafa1f34 dafa1f20 c02f9db4 c0383800
[  234.553494] 1f20: 00000002 dc3b8180 dafa1f64 dafa1f38 c02f9fb8 c02f9d98 00000000 c031c82c
[  234.561712] 1f40: c1504dc8 dc3b8180 00000000 00000000 dc3b8180 b6f10aec dafa1fa4 dafa1f68
[  234.569929] 1f60: c02fa230 c02f9f10 00000000 00000000 00000002 322c3fe2 00000005 00000001
[  234.578146] 1f80: 0000000a 00000000 00000004 c0108f44 dafa0000 00000004 00000000 dafa1fa8
[  234.586362] 1fa0: c0108d40 c02fa1e0 00000001 0000000a 00000003 b6f10aec 00000002 b6f31620
[  234.594578] 1fc0: 00000001 0000000a 00000000 00000004 beb97518 00000005 00000006 00000007
[  234.602795] 1fe0: 00000000 beb9745c b6ef2c8d b6ed23b6 00070030 00000003 00000000 00000000
[  234.611029] [<c075dd88>] (ehrpwm_pwm_config) from [<c075b3b8>] (pwm_apply_state+0x120/0x1c4)
[  234.619511] [<c075b3b8>] (pwm_apply_state) from [<c075bb08>] (duty_cycle_store+0x7c/0xb0)
[  234.627740] [<c075bb08>] (duty_cycle_store) from [<c086f318>] (dev_attr_store+0x28/0x34)
[  234.635881] [<c086f318>] (dev_attr_store) from [<c03842a0>] (sysfs_kf_write+0x50/0x54)
[  234.643840] [<c03842a0>] (sysfs_kf_write) from [<c03838ec>] (kernfs_fop_write+0xf8/0x1d4)
[  234.652068] [<c03838ec>] (kernfs_fop_write) from [<c02f9db4>] (__vfs_write+0x28/0x48)
[  234.659939] [<c02f9db4>] (__vfs_write) from [<c02f9fb8>] (vfs_write+0xb4/0x1c0)
[  234.667287] [<c02f9fb8>] (vfs_write) from [<c02fa230>] (SyS_write+0x5c/0xbc)
[  234.674381] [<c02fa230>] (SyS_write) from [<c0108d40>] (ret_fast_syscall+0x0/0x54)
[  234.681993] Code: e5980000 e6ffa07e eb0494a7 e5985028 (e1d570b0)
[  234.688119] ---[ end trace be0b29146c9dec19 ]---
debian@beaglebone:/var/lib/cloud9/8ece4ad83ac67e5215af6a3d686d0ae2$
```