beagleboard/librobotcontrol #155: pwm

guoguowuxuan, 2019-03-22:
**Describe the bug**
  when I run rc_test_sevos.c,it interrupt by "Bus error",Cross-compilation,find when run rc_servo_init()   shared_mem_32bit_ptr[i-1] = 42，it occur “Child terminated with signal = 0x7 (SIGBUS)”
**Platform information**
beagle bone blue ，Debian 9.5 2018-10-07 4GB SD LXQT 
PC：Ubuntu linux 16.04,eclipse CTD，arm-linux-gnueabihf
Please let us know the board and kernel version. On BeagleBoard the date of the running image is helpful too.

StrawsonDesign, 2019-03-22:
Did you compile the library yourself or are you using the librobotcontrol Debian package that comes installed in the Debian image? It seems that the PRU subsystem isn't starting which likely means the device tree has been modified to not include the PRU or the PRU binaries didn't get compiled and installed. The 2018-10-07 image has the full Debian package installed which includes library, PRU binaries, and rc_test_servos example preinstalled and working, so something must have been modified from that state.

guoguowuxuan, 2019-03-22:
I compiled the library in my PC and running Executable Files in the beagle bone blue. how can include the PRU binaries when i  compiled in my PC?

guoguowuxuan, 2019-03-22:
I compiled the library in my PC and running Executable Files in the beagle bone blue. how can include the PRU binaries in when i compiled by Eclipse CTD in my PC?

StrawsonDesign, 2019-03-26:
They are installed as part of the deb package which should already be installed in your 2018-10-07 debian image. Can you please post the result of the following commands on beaglebone?

apt list librobotcontrol
ls /lib/firmware/ | grep "pru"

May I ask why you are compiling everything yourself? Also why are you using a cross compiler?

guoguowuxuan, 2019-03-27:
thank you very much, I find my debian image is boot from SD,the pru and adc driver is not load.now,the problem is solved.the eclipse is easy to use,i can find the problem quickly by debug.