beagleboard/librobotcontrol #156: Specifications

HugoJansson, 2019-03-22:
I can´t find engine specifications anywhere

jadonk, 2019-03-26:
Are you talking about the motors as part of the EduMIP or the motor drivers as part of BeagleBone Blue or Robotics Cape?

The motor driver strength is now listed in the FAQ for BeagleBone Blue:
https://github.com/beagleboard/beaglebone-blue/wiki/Frequently-Asked-Questions-(FAQ)#What_is_the_DC_motor_drive_strength 

Based on the connectors, the per channel limit is 1A.

HugoJansson, 2019-03-26:
I´m asking about the motors that are a part of the EduMIP. Is it possible to get the engine and encoder specifications?

StrawsonDesign, 2019-03-26:
full motor, encoder, and frame specifications are here:
https://github.com/StrawsonDesign/embedded_control_and_robotics_assignments/blob/master/Assingment%205%20-%20Balancing%20EduMiP.pdf

closed 2019-03-26 by StrawsonDesign