beagleboard/librobotcontrol #159: rc_battery_monitor reports 4 cells with nothing plugged in

dmalawey, 2019-04-09:
**Describe the bug**
rc_battery_monitor reports the wrong number of cells.  

**To Reproduce**
Steps to reproduce the behavior:
1. run "sudo rc_battery_monitor" at any time with approximately 12.0v plugged into the barrel plug on the beaglebone.  No battery is plugged into the JST connector
2. the output says: \
2S Pack   Jack   #Cells   Cell\
 8.05V   12.25V  4       3.06V\

**Expected behavior**
The program will report "zero cells" or report "undetermined".

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Platform information**
 Debian Image 2018-10-07

StrawsonDesign, 2019-04-23:
This is the intended behavior to support running the cape/blue off of a LiPo battery on the DC Jack. The white JST connector can only physically connect a 2S pack, but 2,3,&4S packs can be connected to the DC jack.

dmalawey, 2019-04-24:
In the above case, my 3 cell li-ion is connected to the DC jack and is being reported as a 4-cell with low voltage (2.81v) instead of a 3-cell with good voltage (3.74v) per cell.  No battery is connected to the white JST connector.